Ultimaker/Cura #7720: Scaling, position, rotation menus always visible.

st4szek, 2020-05-08:
Please create a "pin on/always on" function for left-side menus for scaling, rotation, size and so on... It is annoying for me that it disappears when I miss the field to write the value. Then the model unselects and the menu disappears. Problem is that you must hit directly in the text field or checkbox field. Please improve here.

Ghostkeeper, 2020-05-12:
I consider it a bug that the selection changes if you click on empty areas of that panel.

st4szek, 2020-05-12:
Yes, that's right. When you click out of the table the part unselects, and opened window for model settings disappears. It happens even you click inside this window with settings but out of the text field or checkbox.

nallath, 2020-05-13:
That was an easy fix :) It's in the next release (4.7)

commit 8aabf113 (2020-05-13): Catch mouse events in toolpanel

Fixes #7720
closed 2020-05-13 by nallath