Ultimaker/Cura #7721: "Connect Support Zig-Zags" always connects them, checked or unchecked

KenLucke, 2020-05-09:
**Application version**
4.6.1

**Platform**
Mac OS 10.154

**Printer**
Creality CR-10v2 (derivative of CR-10s

**Reproduction steps**
- Create Supports with Zig-Zag pattern
- Uncheck "Connect Support Zig-Zags"
- Slice

**Screenshot(s)**
For some reason, gitHub REFUSES to accept any of my files (Amazon is the same way, but every other site works fine)
Here is a Dropbox link to the sccreenshot:
https://www.dropbox.com/s/w2rct8hgdb9mcxk/Ultimaker%20Cura%204.6.1%202020-05-08%2018-56-08.jpg

**Actual results**
Supports are created with zig-zags connected.  Versions previous to 4.6.x worked fine

**Expected results**
Supports should not have zig-zags connected

**Project file**
https://www.dropbox.com/s/f9enl9qgc41f0yz/Crab%20Flag%20Drop%20Weight.3mf

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

smartavionics, 2020-05-09:
You need to use the "lines" pattern, not zigzag?

KenLucke, 2020-05-09:
> You need to use the "lines" pattern, not zigzag?

A)  I don't want to use "lines," I want to use zig-zag, as I almost always do and have
B) Begs the issue of the checkbox status being ignored.

smartavionics, 2020-05-09:
AFAIK, the connect support zig-zags setting doesn't affect zig-zag pattern, it affects the lines pattern (and other pattern when using the similar setting for infill). It's not meant to affect the zig-zag pattern.

KenLucke, 2020-05-09:
> AFAIK, the connect support zig-zags setting doesn't affect zig-zag pattern, it affects the lines pattern (and other pattern when using the similar setting for infill). It's not meant to affect the zig-zag pattern.

Then explain why
a) It is called "Connect Zig-Zag Supports, and,
b) it only appears as an option when Zig-Zag supports are chosen.

smartavionics, 2020-05-09:
I do not know the answers to any of those questions, sorry.

KenLucke, 2020-05-09:
Here is a link to a view previously sliced file with "Connect Support Zig-Zags" unchecked.  It was:
`;Generated with Cura_SteamEngine 4.5.0`
according to the header in the gcode file

https://www.dropbox.com/s/l2buxkpvejof47w/Mesh_Z1_Y1_X2%20PLA%2010%25%20210%C2%BA%7C65%C2%BA_multi%20-%20Ultimaker%20Cura%204.6.1%202020-05-09%2015-46-10.jpg

This shows the _expected_ (and previous to v4.6.x) results

I[ wish I could figure out why github and amazon won't accept my jpgs, pngs, gifs, or any other file, when every other site has no problems with them.  It would make life so much easier here.]

smartavionics, 2020-05-10:
Having found time to play with this, I see that:

1 - connect support zig-zags is visible for zigzag support but has no effect, connections are always made.

2 - that setting is not visible for lines pattern and makes no difference whether set or not, no connections are ever made.

3 - that setting is visible for some of the other support patterns and appears to be effective in controlling the connections.

This behaviour differs from the behaviour of the similar setting used to control the connecting lines in infill. There, the setting is not visible when printing zig-zag lines and is visible for most of the other patterns and it is effective when using the lines pattern.

IMHO, it would be better if the behaviour of the setting for the support infill matched the behaviour of the setting for infill.

smartavionics, 2020-05-10:
Ah, I realise that there are two settings here, connect support lines and connect support zigzags. That's very confusing. Think I'll just shut up now...

Ghostkeeper, 2020-05-12:
See my article on the [Connect Support ZigZags](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/support_connect_zigzags.md) setting in the Settings Guide.

Connect Support ZigZags makes the pattern connect back onto itself at each endpoint. It's the difference between this (off):
[image]
and this (on):
[image]

I loaded up your project file and it works for that one too, except that you have an extra Support Wall so it's not very visible. If you disable Connect Support ZigZags you'll get one line that ends in nothing here:
[image]
And one on the other side as well.

What you really mean to do is to set the [Support Wall Line Count](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/support_wall_count.md) to 0 instead of 1.

closed 2020-05-12 by Ghostkeeper