Ultimaker/Cura #7722: Raft does not take flow/initial layer flow into account

CleyFaye, 2020-05-09:
**Application version**
v4.6.1

**Platform**
Windows

**Printer**
Creality Ender-3

**Reproduction steps**
1. Set flow and initial layer flow to 100
2. Slice model
3. Change flow and initial layer flow to 80
4. Slice model again
5. See that gcode for raft ignore flow

**Screenshot(s)**
[image]
You can see that the part before Layer 0 is the same in both files, but right at layer 0 it starts to differentiate because of the different flow values.

**Actual results**
The raft code seems to completely ignore any flow setting

**Expected results**
Flow should be used for raft extrusion

**Project file**
-

**Log file**
-

**Additional information**
Most "bed adhesion" methods have a separate "flow" setting; maybe raft is just missing one.

Ghostkeeper, 2020-05-12:
Raft is a bit of a weird one. It just has a Raft Base/Middle/Top Line Width which is essentially your flow. This line width doesn't affect the spacing between walls in the case of raft.

Here is a diagram of how various settings affect the dimensions in a raft:
[image]

CleyFaye, 2020-05-12:
I see. By playing with the various line width I can effectively control how much material is extruded by each layer. It's not a bug then.

Maybe having a single setting to tune that behavior could be seen as an enhancement then, unless it's not that obvious (I can see how scaling all the lines with the same ratio could cause issue with the different heights).

Ghostkeeper, 2020-05-14:
It's just a bit inconsistent with the rest. In general the raft settings are perhaps too detailed, which makes it hard to use. But that is necessary to get a good raft with some difficult materials. I think we'll leave it that way, sorry.

closed 2020-05-14 by Ghostkeeper