Ultimaker/Cura #7724: Automatic Sacrificial layer generation

perikanttila, 2020-05-09:
When I model a cad file for printing I will add sacrificial layer to the holes that require otherwise support, so the holes with it looks say 0.2 thick layer closing it.

This would be very handy in inset holes eg. for socket cap screws etc.

Could Cura automatically create such bridging layer with the current layer height.

I have added pictures to illustrate how I use sacrificial layers in my parts

[image]
[image]
[image]

smartavionics, 2020-05-09:
Hi, I use that technique myself and agree it would be neat if Cura could add those single layers automatically. I will give the suggestion some thought.

smartavionics, 2020-05-10:
Hello again, I have done some work on this and have an initial implementation. I added a Remove Holes Under Walls setting that enables this behaviour...

[image]

[image]

[image]

perikanttila, 2020-05-10:
nice!

...
Kari

perikanttila, 2020-09-03:
Is this feature coming to which release, not found in 4.7

Ghostkeeper, 2020-09-07:
It is not planned for a specific release yet. We have this on our backlog at the moment. Actually, it is considered a duplicate of #7324.

closed 2020-09-07 by Ghostkeeper

perikanttila, 2020-09-07:
thanks, but #7324 is first layer but this is not the same as sacrificial layer on holes that has support roof before the smaller part of the hole continues. 

...
Kari

Ghostkeeper, 2020-09-08:
Yeah, true. In our backlog we've written it down as being generic for the first layer as well as later layers though.