Ultimaker/Cura #7727: Numpad keys are ignored.

elf128, 2020-05-09:
**Application version**
4.5.0

**Platform**
Linux, KUbuntu 18.10

**Printer**
n/a

**Reproduction steps**
1. Click on any editable property. Could be any edit field in the program settings or drop-down panel with print settings.
2. Try to type any number using Numpad keys.
3. Try to press NumLock. And type more numbers. 
5. Notice, that behavior didn't change.   

**Actual results**
State of NumLock is 100% ignored and keys always behave like NumLock is OFF.

**Expected results**
Numbers are entered correctly.

**Project file**
n/a. The issue occur even in empty window.

**Log file**
n/a. Key presses are not logged.

**Additional information**
It's laptop. I can use it's own keyboard or external USB keyboard.
The issue occurs in both cases.

nallath, 2020-05-12:
I doubt that this is an issue on the side of Cura. I also just tried this on my machine and it works just fine. Do you also have this issue with other software?

Ghostkeeper, 2020-05-12:
Works fine on Ubuntu 20.04 for me as well. This is pretty much always how I type solitary numbers.

closed 2020-06-19 by no-response[bot]

spontarelliam, 2022-02-10:
I have this same problem on Debian 10. Numpad does not work.