Ultimaker/Cura #7730: versie 4.6 and 4.6.1 when using ender 3 pro

COIScois, 2020-05-10:
With version 4.6 there is issue when printing with the Ender 3d pro. Cura is running on a Mac.
With the new version the machine stop printing after , about , a hour printing. 
the machine stops in the middle of the program. By all the products i did try.

It happens over en over again, at the same spot. It looks to me something in the postprocessor. 
Using micro-sd, usb cable or Octoprint it give the same issue.

When i go back to version 4.5.0, there is no problem at all.

mahtDFR, 2020-05-11:
This machine profile is not maintained or tested by us.
Here is a [list of PRs](https://github.com/Ultimaker/Cura/pulls?q=is%3Apr+ender+3+) that have been submitted. We will pull those into the code, and take it on good faith that each profile is properly tested and works. We don't have Ender 3 equipment so we do no testing from our side.

The best thing to do is to contact the profile author to see if they can make a new PR to fix breaking changes between versions.

fieldOfView, 2020-05-11:
> This machine profile is not maintained or tested by us.

That is all nice and dandy, but the issue at hand is not about the ender 3 pro definition. There is nothing in a printer definition that makes the printer stop "after about an hour, somewhere in the middle of a layer". 

Something seems to be corrupt in the gcode you are trying to print. Please post a link to a gcode file and we can see if there is something that would explain why the print stops after about an hour.

Ghostkeeper, 2020-05-12:
I have a few questions before we can debug this.

1. Is layer view displaying the whole print?
2. Where in the print is it stopping? What does the g-code say at that point in the file?
3. Could you supply a project file?

closed 2020-06-19 by no-response[bot]