Ultimaker/Cura #7732: Cura 4.6.1 custom gcode wont execute

00ihsan, 2020-05-10:
**Application version**
(The version of the application this issue occurs with.)
4.6.1

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Win 10 x64

**Printer**
(Which printer was selected in Cura?)
Ender 3 Pro

**Reproduction steps**
1. (Something you did.)
I added custom gcode OCTO100 to the start gcode
2. (Something you did next.)
Sliced and uploaded to octoprint. 

3. The octo100 command did not execute. 

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)
The OCTO100 command dit not excecute. 

**Expected results**
(What should happen after the above steps have been followed.)
The OCTO100 command to be executed. 
**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

00ihsan, 2020-05-10:
[cura.log](https://github.com/Ultimaker/Cura/files/4606725/cura.log)

closed 2020-05-10 by 00ihsan

fieldOfView, 2020-05-11:
Did you mean to close this issue?