Ultimaker/Cura #7733: RETRACTION DISTANCE CURA 4.6 WITH BONDTECH EXTRUDER

TONYDG, 2020-05-11:
I have change the original extruders to the Bondtech ones. And it was working grate, until I change to Cura 4.6. The retraction distance became very erratic to the point that in the part I was printing the PLA filament broke between the extruder and the hot end.
Then I switch back to Cura 4.5 and did the same part with the same stl. and evrything work great again.
I would love to here anyone on how to solve that.
Thank you
Antonio
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

mahtDFR, 2020-05-11:
Thanks for including the issue template, but can you fill it out please? Some project files and logs here would also help us to help you more.

ungutknut, 2020-05-26:
What is your retraction speed? If you operate the bondtech extruder with retraction speeds of an extruder with 1:1 gear ratio it will skip steps, causing erratic retractions.

And please also upload the project files for your print with cura 4.5 and 4.6.

closed 2020-06-19 by no-response[bot]