Ultimaker/Cura #7734: change the first gcode send

00ihsan, 2020-05-11:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

i use octoprint with the plugin gcode system command. i made a gcode that turns on the priner. but it wont work because the bed starts heating before the gcode for turning the printer on is send. so the printer gets into thermal runawaymode before the code is send.

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

a textbox where i can set the start gcode. i know there is a kind of simmilar box. dut as i said that waits first for the bed to warm up. the textbox i want has to be executed before any other gcode.

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

a scriptthat does it for me.

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

everyone using octoprint.

**Additional context**
(Add any other context or screenshots about the feature request here.)
[image] before

[image] after

fieldOfView, 2020-05-11:
Use a Search and Replace post processing script.
Search `;Generated with`
Replace `OCTO100 ; Generated with`

00ihsan, 2020-05-11:
I don’t know how to make such a script.

fieldOfView, 2020-05-11:
You don't have to make it, you can just "add" it.
See Extensions -> Post processsing -> Modify gcode".
Press "add a script" and select "Search and Replace"

00ihsan, 2020-05-11:
👍

closed 2020-05-11 by 00ihsan