Ultimaker/Cura #7736: Cura keeps crashing after install

theSkyKing30, 2020-05-11:
**Application version**
(The version of the application this issue occurs with.)
version 4.6.1

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 pro 64 bit 
**Printer**
(Which printer was selected in Cura?)
Ender 3 pro 

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
When I download Cura it works until I close the application. As soon as I try to reload the application the program crashes and fails to load. I've tried running the program as an administrator, used the compatibility mode troubleshooter, tried uninstalling and reinstalling and I still get the same result. I also reached out to tech support and followed all of their protocols (which were what I did initially anyways) and even they weren't able to help me. The program runs just fine on my laptop, but for some reason it will not run on my desktop. I've attached my DxDiag so hopefully someone can figure out what's going on. 

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)
Cura crashes after I close the program for the first time after install. 
**Expected results**
(What should happen after the above steps have been followed.)
Cura should continue to run after closing the program. I shouldn't have to keep reinstalling the software in order to get it to run properly. 
**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/4607447/DxDiag.txt)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-05-11:
Can you try starting Cura with only one display enabled? If that works, go to the preferences and untick the "Restore window position on start" checkbox. Chances are the application works with both displays enabled and that preference disabled.

theSkyKing30, 2020-05-11:
Thank you! That fixed it!

closed 2020-05-12 by nallath