Ultimaker/Cura #7737: Option to add name of slice profile to job name

joeySeal, 2020-05-11:
Currently in Preferences > General user can check "Add machine prefix to job name" it would be useful if the user could also check a box that would add the current slicer profile name to the print job name. An example file name would be TAZ6_PLA-FAST_benchy.gcode <printer_name>_<profile_name>_<model_name>.gcode

mahtDFR, 2020-05-11:
One solution: you can use the [Gcode filename format](https://marketplace.ultimaker.com/app/cura/plugins/rgomezjnr/GcodeFilenameFormat) plugin. This allows you to manipulate the gcode filename output.

joeySeal, 2020-05-11:
Thanks, I am trying that now. Is there a way to use the profile name an option for the file name? It seems like the options are all more specific such as material type.

mahtDFR, 2020-05-11:
I just checked the contributor portal and it seems there's an update from the author for this that does exactly that! I'm checking it now. If you can hang on a day or so it will be live.

joeySeal, 2020-05-11:
awesome thanks

mahtDFR, 2020-05-11:
Here's a preview (using an Ultimaker 3)
[image]

Ghostkeeper, 2020-05-13:
This was once an idea for a feature request. But that also didn't get through the planning, in the end. Now that there is a plug-in it's not likely to get much priority to get it built-in natively, so I'll mark this as deferred.

rgomezjnr, 2020-05-14:
Hi @joeySeal,

I'm the author of Gcode Filename Format. I did add support to the plugin for the profile name in version 1.3.1. You can get it from the Cura Marketplace.

For example, the following format:

`[base_name] [profile_name] [brand] [material] lw [line_width]mm lh [layer_height]mm if [infill_sparse_density]% ext1 [material_print_temperature]C bed [material_bed_temperature]C`

Will result in the following filename:

`PI3MK3M_paperclip Draft Generic PLA lw 0.8mm lh 0.2mm if 20% ext1 200C bed 60C.gcode`

Let me know if you have any questions or feedback.

Thank you @mahtDFR for testing and approving the plugin, and for the suggestion to joey.

joeySeal, 2020-05-14:
I just updated to 1.3.1 and it works as expected thanks!

closed 2020-05-14 by joeySeal