Ultimaker/Cura #7738: spire support when support infill layer thickness is larger than layer height

elsbiet, 2020-05-11:
**Application version**
4.6.1

**Platform**
linux

**Printer**
self built
**Reproduction steps**
- create an model with a spire pointing downwards. 
- alter print settings: set 'support infill layer thickness' to the double of the layer thickness (.4mm in my case)
- slize the model
- observe the support of the above spire
in my case 2 support layers are missing which made my print fail

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

mahtDFR, 2020-05-12:
What's 'spire support'? Can you provide some screenshots and 
save a project file (File -> Save...), zip up the .3mf file and attach to this issue so I can try to reproduce it with the same settings as you.

elsbiet, 2020-05-17:
i meant support for a downward pointing spire.

question: does cura has a way to cut a cuboid shaped part out of my slized model which i could send to you? (i'm not supposed to publish the complete model)

Ghostkeeper, 2020-05-19:
No, Cura can't export only part of a model.

Perhaps you meant how it supports something like this? 
[image]

closed 2020-06-19 by no-response[bot]