Ultimaker/Cura #7739: Add printer by IP: only one network plugin can be enabled at the time

loociano, 2020-05-11:
**Application version**
Cura 4.6.1

**Platform**
Windows 10

**Printer**
N/A

**Reproduction steps**
1. Add Printer
2. Add printer by IP
3. Enter printer IP address
4. Add

**Screenshot(s)**
N/A 

**Actual results**
Cannot connect to network printer if two network plugins are enabled. Only `UM3NetworkPrinter` plugin issues HTTP requests.

**Expected results**
Method `addManualDevice()` is invoked on all enabled network plugins.

**Project file**
N/A

**Log file**
N/A

**Additional information**
Hi! I am unsure whether this issue is a bug or a feature request, feel free to tag.

I am developing a network plugin for a 3rd party network-enabled printer, and noticed that, when trying to add a printer manually by IP address, two network plugins cannot operate at the same time as of Cura 4.6.1.

Method `checkManualDevice` in `DiscoveredPrintersModel` only selects the top network plugin based on `PRIORITY`:
https://github.com/Ultimaker/Cura/blob/fe27da5e8a945d8b79834aa5779b323045f7e742/cura/Machines/Models/DiscoveredPrintersModel.py#L164

Setting `ManualDeviceAdditionAttempt.PRIORITY` in a second network plugin does not guarantee that it is chosen by Cura. In fact, `UM3NetworkPrinter` plugin always prevails in my testing.

Workaround is to disable `UM3NetworkPrinter` plugin to let the second network plugin operate, although this is not ideal.

nallath, 2020-05-12:
Yeah, you're right. It shouldn't work like that. All plugins should try and see if they can (manually) connect to that printer (and based on if it succeeds, a discovered printer should be added).

loociano, 2020-05-12:
Thanks for confirming Jaime.

PR #7782 (2020-05-25): Add printer by IP: support more than one network plugin at the time
This suggested change fixes #7739.

Tested on `master` with a second network plugin (WIP) enabled. Both plugins attempted to communicate with the printer at the specified IP address:

```
cura.Machines.Models.DiscoveredPrintersModel._attemptToAddManualDevice [171]: Network plugin UM3NetworkPrinting: attempting to add manual device with address 192.168.0.136.
UM3NetworkPrinting.src.Network.LocalClusterOutputDeviceManager.<lambda> [71]: Error transferring http://192.168.0.136/api/v1/system - server replied: Not Found
cura.Machin [... cut]
closed 2020-05-25 by nallath