Ultimaker/Cura #7740: Overrides Max Speed to 8mm/s (M203)    V4.6.1 Win10 Creality CR-10 template w/ RepRap

magnokneeto, 2020-05-12:
Problem summary
Cura 4.6.1 creates a few lines of code at the beginning of the G-code file that over-ride machine settings and one is very much incorrect and sets the max machine speed to 8mm/s
(also there is no way to toggle back to a solid view after slicing.  Only X-ray and layer)

Cura V4.6.1

Win10

Creality CR10 w RepRap 

**Reproduction steps**
1. Compiled STL to G-Code

Creates a machine setting override sequence in the header section that was not on Version 3.5.

Generates line: 
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max federate

My Duet2 OEM settings (RepRap) are :
M203 X9000.00 Y9000.00 Z180.00 E6000.00 ; set maximum speeds (mm/min) 

Both M201 and M204 commands which show up just before and after it 
are specified as :
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration

My default is: M201 X500.00 Y500.00 Z100.00 E5000.00   ; set accelerations (mm/s^2)

This has occurred with multiple slicing to g-code translations and has been on every file I've converted since I updated to Cura 4.6.1 this morning.

**Screenshot(s)**
The bad line of code is line 18 of the G-code file.  I attached a text file showing the g-code header.

**Actual results**
This sets my max speed at approximately 8mm/s (500mm/min*1min/60s)

**Expected results**
These machine overrides should not exist.  If my machine is setup correctly, I don't need random software over-riding it in the background
[Cura Max speed overide glitch.txt](https://github.com/Ultimaker/Cura/files/4613188/Cura.Max.speed.overide.glitch.txt)

nallath, 2020-05-12:
Well, these g-codes are as provided by us by the community. If you want to change your start / end g-code, you can do that via:
preferences-> Printers -> machine settings. 

That should open a window where you can change the start g-code. 

By default, creality uses marlin, so it doesn't set the values in mm/m but in mm/s (https://marlinfw.org/docs/gcode/M203.html).

closed 2020-05-12 by nallath

magnokneeto, 2020-05-12:
Thank you!
This solves my issues by being able to override my settings.  However, for new users this will remain a problem and it is not related to the mm/min vs mm/s marlin vs RepRap standards.  500mm/s is excessive.  9000mm/min works out to 150mm/s.

Cheers

Ghostkeeper, 2020-05-14:
Yes, but for these printers we don't know what works and what doesn't. We can't test anything without having the printer hardware, and don't have the time to finetune all 260 printer profiles either, so we need to rely on the users of that printer to provide pull requests to fix their printer defaults.

magnokneeto, 2020-05-14:
I really appreciate this software, it works great. And I also appreciate that with 260 printers it is difficult, if not impossible to verify all settings. 
With regards to this problem :1.  I gave you a fix, a simple "gotcha, we'll update." along with the advice on how to remove that line of code from my current situation which you provided would have been enough. 
2. If you are going to wash your hands of any responsibility and blame it on cloud input and leave glitches, why not leave them out?  Rep rap does a pretty darn good job optimizing characteristics for their system to integrate with various configurations.  Why do the work twice and undue what the "experts" recommend?
I'm not trying to be a pill but these responses leave me 🤦‍♀️🤦‍♀️🤦‍♀️. 
Cheers

Ghostkeeper, 2020-05-18:
> I gave you a fix, a simple "gotcha, we'll update." along with the advice on how to remove that line of code from my current situation which you provided would have been enough. 

We normally do that if the bug is obviously a bug. In this case though, the author of that definition explicitly wrote that down and documented the reason for it. It's your word against theirs. The Creality CR10 has a lot of users and has been in Cura for a long time, and you mention that you've customised the firmware for it.

> If you are going to wash your hands of any responsibility and blame it on butt input and leave glitches, why not leave them out?  Rep rap does a pretty darn good job optimizing characteristics for their system to integrate with various configurations.  Why do the work twice and undue what the "experts" recommend?

Perhaps, but Creality doesn't ship with RepRap, does it?

People need customised start g-code because the firmware expects the g-code to set up the print at the beginning (e.g. with the K factor for linear advance) or because they modified the printer themselves (like you).

People need to override the speeds, acceleration and jerk because the optimal speed/acceleration/jerk depends on the sort of speeds you are going to print the various structures at, and the shapes you're making the print head follow. There is no one-size-fits-all for speed/acceleration/jerk, even if you're not taking the different quality profiles into account.