Ultimaker/Cura #7741: FR: Custom Colors in preview (for dual extrusion), and preview retractions.

chepo92, 2020-05-12:
**Is your feature request related to a problem? Please describe.**
In dual printing preview, it doesn't show different colors for different exruders. 

**Describe the solution you'd like**
It would be great that you can customize not only the extruder color but all of the features, also a way to display retractions (Simpli 3D does). Maybe just a right click > set color..., a field where you have to input the rgb/hex code (also a palette would be nice but maybe too much), a button to set and other to default, and cancel . 

**Describe alternatives you've considered**
none

**Affected users and/or printers**
Anyone with dual or more extruders

**Additional context**
Related to:
https://github.com/Ultimaker/Cura/issues/6040

mahtDFR, 2020-05-12:
> In dual printing preview, it doesn't show different colors for different exruders.

You can if you're reviewing material color

[image]

As mentioned in the related ticket:

> Theming should take care of the rest. I don't see the need for the user being able to customise that himself, other than through the theme system.

I think we're reluctant to introduce too many random colours beyond the existing themes. But perhaps there is some value to showing retractions. What's the use case for this?

DkOuvE, 2020-05-14:
+1 for showing retractions to help with tuning settings related to coasting, combing and travel.

Ghostkeeper, 2020-05-15:
Retractions are shown with a lilac line rather than a deep blue line.

DkOuvE, 2020-05-15:
Helpful information to know about.

vaughnsc, 2020-07-12:
Just browsing, but thought it might be useful to point out that some people (myself included) have anomalous color-perception commonly lumped together as '[color-blindness](https://en.wikipedia.org/wiki/Color_blindness#Classification),' that make it very hard to see a shift in green hues or from blue to violet (red component) even if seem glaringly obvious to normal color vision. Its a strong argument for letting the user pick the colors (whatever works for them.)

chepo92, 2020-07-12:
> Just browsing, but thought it might be useful to point out that some people (myself included) have anomalous color-perception commonly lumped together as [...]

thats a good argument, indeed I find the keep colors just for legacy reason a bit weak, I'm not color blind but sometimes orange red and yellow are difficult to differentiate as there are no much color/contrast difference. I haven't had a look into code for this but it should straight forward to hardcode new colors, also a click and set rgb hex color in the gui should be not so difficult to add as feature.

nallath, 2020-07-13:
Have you checked the special themes for color blindness? Do those solve the problem for you?