Ultimaker/Cura #7742: Failure in G-code

jaapfijn, 2020-05-12:
version 4.6.1

During the start of the print the printer flushes the nozzle first and than goes in the X and Y axis to the starting point of the product and then the Z axis to the building plate. When I use the "Z hop when retracted" it makes a faillure. After flussing the nozzle the first move is in Z axis toward the building plate chrashing into the material that was flushed. 

The workarround I use is to open the G-code file and modify it.

;FLAVOR:UltiGCode
;TIME:4334
;MATERIAL:6290
;MATERIAL2:0
;NOZZLE_DIAMETER:0.4
;MINX:81.471
;MINY:96.823
;MINZ:0.27
;MAXX:141.529
;MAXY:126.289
;MAXZ:30.67
;Generated with Cura_SteamEngine 4.6.1
M82 ;absolute extrusion mode

G92 E0
G92 E0
G10
;LAYER_COUNT:197
;LAYER:0
M107
G1 F600 Z1.27
G0 F4500 X89.213 Y97.416 Z1.27
;TYPE:SKIRT
G1 F600 Z0.27
G11
G1 F1800 X91.865 Y97.994 E0.38475
G1 X94.549 Y98.51 E0.77217
G1 X97.24 Y98.95 E1.15868
G1 X99.962 Y99.315 E1.54798
G1 X102.692 Y99.603 E1.93711
G1 X105.408 Y99.813 E2.32325
G1 X108.149 Y99.946 E2.71224
G1 X110.864 Y100.002 E3.09718

I hope you can solve this quickly

Regards,
Jaap

mahtDFR, 2020-05-12:
Sounds like a duplicate of #7382.

Are you able to fill in the issue template just so we have some info about your setup? I've pasted it here:

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

jaapfijn, 2020-05-12:
Application version
4.6.1
Platform
Windows 10 Home 64 bit Surface
 
Printer
Ultimaker 2+
Reproduction steps
1.	Just add Z hop when retracted (hop distance 1mm)
Screenshot(s)
FLAVOR:UltiGCode
;TIME:4334
;MATERIAL:6290
;MATERIAL2:0
;NOZZLE_DIAMETER:0.4
;MINX:81.471
;MINY:96.823
;MINZ:0.27
;MAXX:141.529
;MAXY:126.289
;MAXZ:30.67
;Generated with Cura_SteamEngine 4.6.1
M82 ;absolute extrusion mode
G92 E0
G92 E0
G10
;LAYER_COUNT:197
;LAYER:0
M107
G1 F600 Z1.27
G0 F4500 X89.213 Y97.416 Z1.27
;TYPE:SKIRT
G1 F600 Z0.27
G11
G1 F1800 X91.865 Y97.994 E0.38475
G1 X94.549 Y98.51 E0.77217

Actual results
During the start of the print the printer flushes the nozzle first and than goes in the X and Y axis to the starting point of the product and then the Z axis to the building plate. When I use the "Z hop when retracted" it makes a faillure. After flussing the nozzle the first move is in Z axis toward the building plate chrashing into the material that was flushed
Expected results
first move in X Y direction
Project file
[cura.log](https://github.com/Ultimaker/Cura/files/4614768/cura.log)

see above
Log file
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
Additional information
(Extra information relevant to the issue.)

Ghostkeeper, 2020-05-13:
Yeah, this is a duplicate of https://github.com/Ultimaker/Cura/issues/7382 then. Thanks for the report. We're working on a solution. @nallath is going to the office tomorrow and will test it on an UM2 there.

closed 2020-05-13 by Ghostkeeper