Ultimaker/Cura #7744: Startup Issue 4.6

roupou, 2020-05-12:
**Application version**
4.6

**Platform**
OS: Windows 10
CPU: AMD Ryzen 2400G
GPU: GTX 1050Ti

**Printer**
None Cura does not start

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Splash screen appears and appears to be loading, main screen flashes but immediately closes, icon in tray also dissapears

**Expected results**
Cura should startup.

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4616596/cura.log)
[cura.log](https://github.com/Ultimaker/Cura/files/4616599/cura.log)
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
I read my log files and found the same issue with https://github.com/Ultimaker/Cura/issues/7584.
I unistalled and reinstalled graphics drivers and had the same issue. 
After this i uninstalled the drivers restarted the pc and run cura successfully without installing drivers. While Cura was running i installed drivers. Everything was working but after this i had the same issue again.

astupidmoose, 2020-05-13:
I'm also having this issue, and my logs end at the same place yours does. 

Uninstalling and Reinstalling works, HOWEVER, it only works if during the installation process, I also remove all saved profiles. If I choose to keep any settings or profiles during the uninstall process, same issue. 

I'm also running NVIDIA drivers, GeForce GTX 1070

rburema, 2020-05-13:
Hm, do you also have _this_ in your logs?

```
2020-05-12 18:27:52,485 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: Initialized OpenGL subsystems.
2020-05-12 18:27:52,493 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Version:  4.1.0 NVIDIA 441.87
2020-05-12 18:27:52,501 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Vendor:   NVIDIA Corporation
2020-05-12 18:27:52,509 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: OpenGL Renderer: GeForce GTX 1050 Ti/PCIe/SSE2
2020-05-12 18:27:52,518 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [116]: GLSL Version:    4.0.0
```

Not the discrepancy between OpenGL and glsl! _That shouldn't happen._ We need OpenGL 4.1 & the accompanying glsl 410, this is a mismatch. I've seen this once before (recently even), but then it was fixed by reinstalling the drivers.

roupou, 2020-05-13:
Ok, the problem persists.
the steps i followed.
1) Unistall Cura and manually deleted all files and folders remaining in "Appdata" there was no option for deleting profiles while uninstalling.
2) Unistalled Creality Slicer powered by Cura just in case there is something connecting this with my problem
3) Uninstalled Graphics drivers.
4) Restarted PC
5) Installed Drivers
6) Installed Cura

roupou, 2020-05-13:
> Hm, do you also have _this_ in your logs?
>
> ```
> 2020-05-12 18:27:52,485 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: Initialized OpenGL subsystems.
> 2020-05-12 18:27:52,493 - DEBUG - [...]
 
Yes i have this lines in my logs, 441.87 is the drivers' version i installed. tried installing 416.36 and got an error.

roupou, 2020-05-13:
>  the installation process, I also remove all saved profiles. If I choose to keep any settings or profiles during the uninstall process, sa

Hey, i don't have any saved profiles. what's the location this files supposed to be?
I tried manually deleting all remaining files from directories "/AppData/Roaming" and "/AppData/Local"

astupidmoose, 2020-05-13:
> Hey, i don't have any saved profiles. what's the location this files supposed to be?
> I tried manually deleting all remaining files from directories "/AppData/Roaming" and "/AppData/Local"

I was asked if I wanted to save custom profiles and such during the uninstall process. This is what I meant. 

> Hm, do you also have _this_ in your logs?
>
> ```
> 2020-05-12 18:27:52,485 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: Initialized OpenGL subsystems.
> 2020-05-12 18:27:52,493 - DEBUG - [...]

Yes, I have the same logs. Here is mine: 

> 2020-05-12 22:41:32,367 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: Initialized OpenGL subsystems.
> 2020-05-12 22:41:32,370 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Version:  4.1.0 NVIDIA 388.57
> 2020-05-12 22:41:32,373 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Vendor:   NVIDIA Corporation
> 2020-05-12 22:41:32,378 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: OpenGL Renderer: GeForce GTX 1070/PCIe/SSE2
> 2020-05-12 22:41:32,380 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [116]: GLSL Version:    4.0.0
> 2020-05-12 22:41:32,384 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\default.shader]...
> 2020-05-12 22:41:32,390 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\selection.shader]...
> 2020-05-12 22:41:32,395 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\select_face.shader]...
> 2020-05-12 22:41:32,400 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\default.shader]...
> 2020-05-12 22:41:32,406 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\composite.shader]...
> 2020-05-12 22:41:32,412 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\overhang.shader]...
> 2020-05-12 22:41:32,421 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\striped.shader]...
> 2020-05-12 22:41:32,427 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\transparent_object.shader]...
> 2020-05-12 22:41:32,433 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\striped.shader]...
> 2020-05-12 22:41:32,442 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\xray.shader]...
> 2020-05-12 22:41:32,447 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\xray_composite.shader]...
> 2020-05-12 22:41:32,453 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\default.shader]...
> 2020-05-12 22:41:32,458 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\grid.shader]...
> 2020-05-12 22:41:32,464 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\platform.shader]...
> 2020-05-12 22:41:32,491 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\xray.shader]...

Ghostkeeper, 2020-05-14:
> Hey, i don't have any saved profiles. what's the location this files supposed to be?
I tried manually deleting all remaining files from directories "/AppData/Roaming" and "/AppData/Local"

Those are the correct locations. When doing a factory reset, take care to delete them while Cura is not running, otherwise it will restore some of those files upon closing or upon interacting with Cura.

However I think here the problem is not really related to the configuration, but more to Cura's interaction with your GPU. NVidia's latest driver is doing something wrong here. Take this documentation from the Khronos website: https://www.khronos.org/opengl/wiki/Core_Language_(GLSL)#OpenGL_and_GLSL_versions , I quote:

> For all versions of OpenGL 3.3 and above, the corresponding GLSL version matches the OpenGL version. So GL 4.1 uses GLSL 4.10.

They literally use 4.1 as the example here. If they say they support OpenGL 4.1 and then say that the GLSL version is 4.0, that's a broken driver.

roupou, 2020-05-14:
> 
> 
> > Hey, i don't have any saved profiles. what's the location this files supposed to be?
> > I tried manually deleting all remaining files from directories "/AppData/Roaming" and "/AppData/Local"
> 
> Those are the correct locations. When doing a factory reset, take care to delete them while Cura is not running, otherwise it will restore some of those files upon closing or upon interacting with Cura.
> 
> However I think here the problem is not really related to the configuration, but more to Cura's interaction with your GPU. NVidia's latest driver is doing something wrong here. Take this documentation from the Khronos website: https://www.khronos.org/opengl/wiki/Core_Language_(GLSL)#OpenGL_and_GLSL_versions , I quote:
> 
> > For all versions of OpenGL 3.3 and above, the corresponding GLSL version matches the OpenGL version. So GL 4.1 uses GLSL 4.10.
> 
> They literally use 4.1 as the example here. If they say they support OpenGL 4.1 and then say that the GLSL version is 4.0, that's a broken driver.
> 
> Not to say that Cura shouldn't work on a broken driver though. Maybe we need to check both of them and revert to the compatibility rendering if the GLSL version is too low.

So, what am i supposed to do?

Ghostkeeper, 2020-05-15:
Not all of my comments are commands for the reporter of a bug to do something. Sometimes I'm just analysing the situation. These threads are meant to discuss these issues, after all, not just to request more information.

roupou, 2020-05-15:
> 
> 
> Not all of my comments are commands for the reporter of a bug to do something. Sometimes I'm just analysing the situation. These threads are meant to discuss these issues, after all, not just to request more information! :)

Oh, okay. I wasn't sure if you proposed something or not. Hmmm, my broken drivers affect the graphics globally, not just Cura, right?

rburema, 2020-05-15:
> Hmmm, my broken drivers affect the graphics globally, not just Cura, right?

They might. It's just the way we depend on OpenGL 4.1 as a minimum specifically. Any program relying on 4.0 or lower as a minimum will just work, and any program relying on 4.2 will 'just not' work (or correctly detect the situation and fall back to legacy). So that limits the number of programs that might fail.

Two other things:
- One of our QA people said they encountered this also and asked whether you had a setup with dual graphics cards (Intel & NVidea), because they solved it by disabling their _Intel_ graphics drivers (exactly _why_ or _how_ that fixed the problem is ... unlcear).
- I'll make a quickfix in a few minutes so that instead of crashing it'll fall back to the legacy drivers for situations like these. The preview mode in particular will look bad, and 'lay flat by face' will not be available, but those problems pale in comparison to not being able to start at all.

We can't do anything else than that since the problem is otherwise outside of Cura.

rburema, 2020-05-15:
Just tried to implement the workaround, but that would conflict with _another_ graphics driver workaround (the one where attempting to delete an OpenGL context would crash the program) :-/

So that's a no-go for the moment. What you _can_ do however, is open up `%appdata%/cura/4.6/cura.cfg` and add the line `opengl_version_detect = force_legacy` under the `[view]` header.