Ultimaker/Cura #7746: Allow to record presets for machine, filament

Shine753, 2020-05-12:
Hello,

I see a huge miss in the parameter handling of Cura (and many other Slicers).

I mean that the proposed parameters for the Machine and for the filaments don't seem realistic to me.
It should be possible to define for each defined "machine" any of the available parameters.
And to be able to do this for each "filament" parameters as well.
As we already can do it for the "quality".

And when the user changes the selected machine or filament, asking the user which parameter he wants to keep: the one defined by default for the machine, the one defined by default for filament, the quality, or keep its currently defined specific value.

For example, the flow control is not necessarily the same on an Ender-3 or on an Ultimaker. As it is not necessarily the same on a new Ender-3 as on a slightly worn Ender-3.

The same parameter is not necessarily the same for a TPU filament as on an ABS one.
Finally, this same parameter can be set differently depending on whether you are printing in 0.1 or 0.6 on TPU, or on ABS.

This also applies, for example, to the number of top layers, the presence or absence of a skirt or the infill percentage, which is set differently depending on whether you are printing soft TPU or PLA, on a heated bed or not, etc...

Having almost all the settings all the time separated from the choosen machine or filament is a nonsense to me.

What do you think about this?

rburema, 2020-05-13:
As far as I understood your request, we _already_ do this, except we don't _ask_ when the parameters are changed _if_ they _haven't been altered by the user_.

Take a look at the 'Printing Temperature' for generic PLA on an Ultimaker S3 verus a Creality Ender-5 Plus for an example ('Fast - 0.2mm' versus 'Standard Quality - 0.2mm').

fieldOfView, 2020-05-13:
Also check out the Material Setting plugin, which lets you configure more settings per material (and, incidentally, per printer per material).

Shine753, 2020-05-13:
> 
> 
> Also check out the Material Setting plugin, which lets you configure more settings per material (and, incidentally, per printer per material).

In fact this plugin starts to do what I was thinking about (but it does not ask the user what he wants to keep). But...
1- These settings should be also available for printers.
2- The most important, is that the settings hierarchy defined in Cura is not realistic for a real world usage.

I mean : ANY setting should be able to be defined by the user with a default value following some hierarchy for a Printer > Filament > Nozzle size > Quality. 

Because when you change any of these, one setting among the others can have to be tweaked, while keeping the others unchanged.

As for now, these settings are mostly independent in Cura, that is not what happens in real life.

rburema, 2020-05-13:
> As for now, these settings are mostly independent in Cura, that is not what happens in real life.

They're not independent, we use formulas to calculate some of them.
For example:
```
[values]
cool_min_speed = 7
material_print_temperature = =default_material_print_temperature + 5
material_initial_print_temperature = =material_print_temperature - 5
material_final_print_temperature = =material_print_temperature - 10
material_standby_temperature = 100
speed_print = 60
speed_layer_0 = =math.ceil(speed_print * 20 / 60)
speed_topbottom = =math.ceil(speed_print * 30 / 60)
speed_wall = =math.ceil(speed_print * 40 / 60)
speed_wall_0 = =math.ceil(speed_wall * 30 / 40)
```
This is form `um3_aa0.4_CPE_Fast_Print.inst.cfg`.

rburema, 2020-05-13:
It's true that those formula's can't be overwritten from _within Cura itself_, but you can certainly make your own material-configs and printer-definitions/variants and put those in the user-appdata folder.

Shine753, 2020-05-13:
> Take a look at the 'Printing Temperature'
> You can manage a lot of these with the profiles and materials tabs in the Preferences menu
> They're not independent, we use formulas to calculate some of them.
> make your own material-configs and printer-definitions/variants

You're talking about a completely different thing.
I speak in terms of hierarchy beetween "hardware", you're talking about formulas beetween settings. 
You take examples from some predefined parameters, where I talk about any parameter available.

Also, for example, if I have juste 1 printer, just 2 nozzles and just 2 filaments, the solution you suggest seems to lead to define a tons of configs :
Printer 1 > Filament 1 > Nozzle 1 > Quality 0.1
Printer 1 > Filament 1 > Nozzle 1 > Quality 0.2
Printer 1 > Filament 1 > Nozzle 1 > Quality 0.3
Printer 1 > Filament 1 > Nozzle 2 > Quality 0.1
Printer 1 > Filament 1 > Nozzle 2 > Quality 0.2
Printer 1 > Filament 1 > Nozzle 2 > Quality 0.3
Printer 1 > Filament 2 > Nozzle 1 > Quality 0.1
Printer 1 > Filament 2 > Nozzle 1 > Quality 0.2
Printer 1 > Filament 2 > Nozzle 1 > Quality 0.3
Printer 1 > Filament 2 > Nozzle 2 > Quality 0.1
Printer 1 > Filament 2 > Nozzle 2 > Quality 0.2
Printer 1 > Filament 2 > Nozzle 2 > Quality 0.3

All of this just to tweak a few parameters. Without changing the others.
If this is done, and you want to change a parameters for Printer 1 that does is not tweaked for the rest, you have to update all of these configs.

What I ask is to be able not to do that, just change the parameter you want for Printer 1, and if not overrriden by one of its descendant, being automaticaly taken into account, or if overriden, being asked which one I want to keep.

rburema, 2020-05-13:
> I speak in terms of hierarchy beetween "hardware", you're talking about formulas beetween settings.

These are not as different as you think they are. What you describe already happens internally, if you where to change the definition file for 'Printer 1'.

Shine753, 2020-05-13:
> These are not as different as you think they are. What you describe already happens internally, if you where to change the definition file for 'Printer 1'.

Well then, how do you do to change within Cura only all the "support" settings without changing the others if you switch from PLA to TPU without touching anything else ?

Or all the "adhesion" settings if you switch from Printer 1 to printer 2 without touching anything else ?

DkOuvE, 2020-05-14:
> Well then, how do you do to change within Cura only all the "support" settings without changing the others if you switch from PLA to TPU without touching anything else ?

> Or all the "adhesion" settings if you switch from Printer 1 to printer 2 without touching anything else ?

Sounds like Material Settings with another level of the hierarchy to further modify the final settings set and a base to sync settings from.

I started using Material Settings a few days ago and I plan to make settings for all my printers. Since the loaded material is saved per material, it kinda addresses your second point.
[image]
The ideal would be a base material (eg Fast.PLA) and then a sub material for a specific printer (Flash Fast.PLA). The nice thing though is that Printer Settings allows you to quickly look back and forth at different settings without lag which at least helps to keep things synced.

I tried to crack this problem about a week ago and the confliction, as you've seen, is that you need 3 variables to calculate a setting: Printer, Material, Quality (and nozzle size) (Flash 0.6 Fast PLA, CR10 0.4 Fine ABS, etc) and the best I could come up with without an infinite amount of modifier layers was being able to reference other profile settings (like you can reference dimensions from other sketches in Fusion 360)

Ghostkeeper, 2020-05-14:
> 2- The most important, is that the settings hierarchy defined in Cura is not realistic for a real world usage.

I'm going to assume that you're talking about what's possible in Cura by writing profiles as text files, rather than what's possible in the GUI. The GUI has other constraints that I'm sure you'll agree with: The user needs to understand what's happening. Allowing the user to set which profile is overriding which other profile, for each setting separately, is a ridiculously complex operation for 100% of the users, rounded down from 101%. So restricting this to only the writers of profiles for printer support allows us to assume that the user is expert enough to understand what a container stack is and what the current order of overrides is.

The profile hierarchy, as documented [here](https://github.com/Ultimaker/Cura/blob/master/docs/profiles/container_stacks.md) is indeed fixed but since the lower-level profile list is specific to the combination of higher-level profiles, you can make any combination of settings for any printer configuration. Maximum power.

For instance, if you think the printing temperature should be defined by the material (material should have highest priority) then you don't override the printing temperature in the quality profiles that are specific to that material, nor in the intent profiles specific to those quality profiles. If you think it should be defined by the quality level, don't override it in the intent profiles. If you think it should be specific to the nozzle except for this one material which does require a higher printing temperature, you don't override it in the materials except for this one material which was the exception.

> Well then, how do you do to change within Cura only all the "support" settings without changing the others if you switch from PLA to TPU without touching anything else ?

So you would specify the overrides for those support settings in the material profile for PLA and TPU, and make sure that they are not overridden in the quality or intent profiles for PLA and TPU.

> The ideal would be a base material (eg Fast.PLA) and then a sub material for a specific printer (Flash Fast.PLA). The nice thing though is that Printer Settings allows you to quickly look back and forth at different settings without lag which at least helps to keep things synced.

We have this. Only, the base material and all of its submaterials are contained in the same file. See for example the file of [Generic PLA (1.75mm)](https://github.com/Ultimaker/fdm_materials/blob/master/generic_pla_175.xml.fdm_material). It contains overrides for specific machines and nozzles. Those are the two types of swappable profiles that are higher in the profile hierarchy, so essentially there is a material specific to a certain machine+nozzle combination. The material you see in the GUI is the one specific to your machine+nozzle.

> I tried to crack this problem about a week ago and the confliction, as you've seen, is that you need 3 variables to calculate a setting: Printer, Material, Quality (and nozzle size) (Flash 0.6 Fast [...]

Getting a setting value is quite a complex procedure at this point! See https://github.com/Ultimaker/Cura/blob/master/docs/profiles/getting_a_setting_value.md

DkOuvE, 2020-05-14:
> We have this. Only, the base material and all of its submaterials are contained in the same file. 

Wow. If the Material Settings plugin could support subprofiles that would reduce a lot of redundancy in printer settings and keep everything synced.