Ultimaker/Cura #7747: Bed Adhesion Skirt Option - Allow user to specify whether the skirt should be drawn from outside-in or inside-out

midnighttofurun, 2020-05-13:
**Is your feature request related to a problem? Please describe.**
Would be good for the user to specify this as a customization to minimize print head travel after printing the skirt.  

**Describe the solution you'd like**
Allow user to specify whether the skirt should be drawn from outside-in or inside-out

**Describe alternatives you've considered**
The Bed Adhesion customization options for the Skirt option are sparse.  There are no other options to consider

**Affected users and/or printers**
All users who use skirt as their preferred Bed Adhesion Option

**Additional context**
(Add any other context or screenshots about the feature request here.)

rburema, 2020-05-13:
Skin is supposed to be printed from outside inwards, not doing so is considered a bug at this moment. Is there any scenario you would want to print in to out for skirt? (If so I can _maybe_ reopen this ticket.)

The same [pull request](https://github.com/Ultimaker/CuraEngine/pull/1231) I mentioned in the other ticket will also fix this for 4.7

closed 2020-05-13 by rburema

midnighttofurun, 2020-05-13:
Hi,

That's what I thought too.  I expected the skirt to be drawing outside-in.
But my prints have all be drawing the skirt inside-out.  If this is a bug
then I'll put in a ticket for that.  It appears all the skirt behaviors I
am seeing are bugs.

Thanks.