Ultimaker/Cura #7749: Vertically oriented circular holes - No Support Needed option

perikanttila, 2020-05-13:
Vertical holes are holes penetrating the walls (not top or bottom layers)

Placing support blockers is tedious work, it would be nice if

either

there is a option to place and size the support blocker automatically
(there might be a plugin possibility for this possible??)

or

there is a option to disable support generation on vertical holes

mahtDFR, 2020-05-13:
Maybe you can illustrate what you mean just for clarity? Do you have a project file you can share? What if you set the support placement to 'touching buildplate'?

perikanttila, 2020-05-13:
In order to be able set automatically support inside the object, like in the picture I need to set placement "everywhere" thus need to block all 
[image]
holes.

closed 2020-05-13 by perikanttila

Ghostkeeper, 2020-05-15:
This is really a case of "I want support here, but not there" and there is no local geometric distinction between the two. This is what support blockers are meant for, really. In your case, you can also solve it by cranking the support overhang angle up to 89°, most likely.