Ultimaker/Cura #7751: Adding a Stock Printer Profile for the Artillery Sidewinder X1

ExecuteAu, 2020-05-13:
The Sidewinder X1 has been out for over a year. Cura is the main profile most users of this printer use. Would be great to see it added so its simpler for new users to get into using cura.

There is a great pre-made profile that alot of users download to use with Cura from Thingiverse. It already has the models of the printer done so it shows up in cura as the print bed. This might be a great starting point for you, please find the link below.

https://www.thingiverse.com/thing:4117423

mahtDFR, 2020-05-13:
Have a look over here: https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura

We don't make third party profiles, so someone in the community with access to this printer will need to make a profile, **test it thoroughly** (because we can't) and then submit it as a pull request.

ExecuteAu, 2020-05-13:
I dont know how to do a pull request. Only signed up do put in this request but the profile I linked works extremely well.

mahtDFR, 2020-05-13:
This has already been requested here, you can review the discussion https://github.com/Ultimaker/Cura/issues/6096

Perhaps someone in the community is willing and able to pick this up, but it's not something we're able to do ourselves. But we are happy to review a machine definition if submitted.

closed 2020-05-13 by mahtDFR