Ultimaker/Cura #7753: UX bug: installed materials do not show in UI, creating a system mismatch / silent error

semiconducted, 2020-05-13:
Materials installed in Cura that cannot be used on the selected machine appear as if the installation failed instead of just showing that the Materials cannot be used on this particular machine.

If the Team Ultimaker & Admin quoted below is correct in how Cura works, this approach violates several usability heuristics, leading to increased support calls and errant bug reports. 

> [On 5/7/2020 at 3:23 AM, ctbeke said:](https://community.ultimaker.com/topic/31670-cura-will-not-sync-with-ultimaker-account/?do=findComment&comment=263645)
> Not all printer types support PETG and thus the material is hidden if such a printer is selected. The material is still installed, just not visible.

My professional UX design advice is to handle this use case differently. 
- Hiding incompatible materials creates a mismatch between the perceived system state and the actual system state, especially for users with only one machine.
- It also sets the user up to experience repeated "silent errors" wherein they install the material, get a "success" confirmation, but then see no system change, leaving them baffled as to what went wrong, even though nothing did.

**Proposed Solution**
I suggest indicating that the material is installed but just not available for this machine. 
- It could be as simple as showing the unavailable materials as greyed out / not selectable. 
- Or these materials could all be put in a separate 'incompatible with this printer' group.
- Or in the user preferences, a (default to off) toggle to "Hide Incompatible Materials".
Or... I could go on and on, but would need to work with dev to understand the options available.

**Benefits**
- Creates a consistent UI across different printers.
- Leverages 'muscle memory', as in, users can quickly jump to an area in the list where a certain Material always appears instead of having to reread through the entire list each time.
- Lets users know exactly what materials they can and cannot use with that machine
- Creates an opportunity to highlight benefits of newer machines or sell upgrade kits
- Reduces support 'calls' and bug reports.

**Affected users and/or printers**
Affects all users, but especially those with only one printer.

**Additional context**
[image]
[image]

rburema, 2020-05-14:
The list is (potentially) _huge_ though. Consider all 1.75mm versus 2.85mm filaments for example. The other suggestions sound doable.

I'll forward this to our UX rep. in any case.

Ghostkeeper, 2020-05-15:
We've discussed this with the team (including UX) this morning, but we decided to not do this for two reasons:

* When you import a material that has a different filament diameter than what your extruder supports, you already get a pop-up warning you that it is not visible.
* Displaying a list of the unsupported materials would be a lot of clutter. Clutter that is not useful for the great majority of users, and could even be confusing as well.

There is still the story of what happens when a material from the Marketplace doesn't apply to your printer, but we intend to solve that in a different way.

closed 2020-05-15 by Ghostkeeper

semiconducted, 2020-05-15:
Thanks for the explanation @Ghostkeeper !

I certainly wasn't suggesting showing physically incompatible diameters or all known filaments out there - just the marketpace-added and custom-made ones. 

Either way, my off-the-cuff "solutions" were meant more to illustrate the problem. I assumed there were more use cases to consider and a better UX solution out there that could leverage existing functionality. As an end-user, my working knowledge of the system is pretty limited. :)

And thank you for detailing those other scenarios. 
Elegant design solutions don't just happen, they take hashing through all the bad ideas first!

Keep up the good work!