Ultimaker/Cura #7754: [4.6.1] MacOS Delete key not removing objects

jbryson3, 2020-05-14:
**Application version**
4.6.1

**Platform**
MacOS Macbook Pro

**Printer**
Ender 3

**Reproduction steps**
1. Add a model
2. Highlight the model
3. Press the "Delete" key

**Actual results**
Nothing happens

**Expected results**
The model disappears.

**Additional information**
Fn + Delete works as a workaround. This bug looks like it was fixed with https://github.com/Ultimaker/Cura/issues/6475 but has regressed since then

mahtDFR, 2020-05-14:
Thanks for the report. I don't have a Mac handy to verify this but I'll see if we can reproduce it!

mahtDFR, 2020-05-14:
So we've managed to reproduce it:

When loading the model, the delete key works. It still works when multiplying and moving a model around. If changing a material/print core, then the delete key stops working.

Anyone else on MacOS had this?

jbryson3, 2020-05-14:
I haven't ever had my "delete" key work without also using the "Fn" button. I open Cura, I add a model, I select it, and I cannot delete using the delete key. I have the "Standard Quality" profile selected, and "Generic PLA 0.4mm Nozzle" selected.

upatamby, 2020-05-14:
I have a similar problem in WIndows 10. If I merge two models that are assigned to different extruders I can't remove the merged model, nor with the delete-key nor with the Clear Build Plate command. Or, as a matter of fact, the model IS removed, but it still appears as being there. As soon as I load another model then the old model disappears.

mahtDFR, 2020-05-15:
> I haven't ever had my "delete" key work without also using the "Fn" button. I open Cura, I add a model, I select it, and I cannot delete using the delete key. I have the "Standard Quality" profile selected, and "Generic PLA 0.4mm Nozzle" selected.

> I have a similar problem in WIndows 10. If I merge two models that are assigned to different extruders I can't remove the merged model, nor with the delete-key nor with the Clear Build Plate [...]

Unfortunately I can't reproduce these issues in Windows 10 in 4.6.1 from my side. Here are my reproduction steps:

- Open model with Ender 3 machine definition
- Press delete key
- Model deletes as expected

or:

- Open dual extrusion models with Ender 3 definition
- Merge models using CTRL + G
- Press delete key
- Model deletes as expected

or:

- Open model with Ender 3 machine definition
- Change material type from Generic PLA to PETG
- Press delete key
- Model deletes as expected

You'll need to share a project file or a log (preferably both) for us to be able to analyse the problems there. I suspect it will be outside the scope of this issue so might be better to open a new one specifically for Windows, as this one is concerned with MacOS.

64Florian, 2020-05-17:
In my case I have the issue if I'm on "prepare" screen and "Slice automatically" enable or disable **on a fresh started instance of CURA** (if I use contextual menu and click on delete the other model are removed with backspace as expected)
-> after the Delete key worked one time delete "work all the time" for other object added after only **if you start slicing**  

@jbryson3  could you disable "Slice automatically" in "Ultimaker cura -> preferences -> general"
 - close and restart your cura
 - drop an object
 - click "slice" button
 - try to delete your object as fast as you can  using delete key -> Will my test 100% deleted 

P.S. :
- I use Tevo tarantula printer 
- contextual menu -> delete work well all the time in my case 
Adding my stl and my workspace (but in this case pretty sure the workspace is useless)
[Archive.zip](https://github.com/Ultimaker/Cura/files/4641197/Archive.zip)

After saving the project, still not able to delete the objet
closing cura, restart and open the project as project -> not able to delete in my case 

iMac 27" / mac OS Sierra / Cura 4.6.1 / "Slice automatically" disable

jbryson3, 2020-05-18:
@64Florian I don't have "Slice automatically" turned on.

I did do some more testing with my issue:
1. Open Cura
- All menu's collapsed
- Creality Ender-3 Pro / Generic PLA / Custom Profile
2. Load File
3. Select the model with left mouse button
4. Press Delete, nothing happens
5. With the model still selected, click on the scale tool
6. Press Delete, the model is deleted
7. Load file
8. Select the model with the left mouse button 
- Scale tool is still selected from last time
9. Press Delete, the model is deleted

I've done some extensive clicking around and trying to delete models after different events in my Cura workflow. Sometimes the models delete, sometimes they don't.  What seems to be a consistent fix for it is switching to a different modification tool (Move to Scale, or Scale to Move) before I press delete.

64Florian, 2020-05-18:
Thank you for your tests,
When I made my tests I always closed Cura to reproduce (because each time I'm able to delete the model using delete key, after I can delete all the time)
As I can read you didn't try to slice your model (in my side it's the solution see as "unlocker" to be able to delete the model using delete key)

I noticed, for me I use drag and drop to add the model in Cura (but doesn't sem to change anything)

I tested for selecting "scale tool" end I reproduce as you describe -> the delete key work after 
"rotate tool" work as the same way  -> the delete key work after 
I have test to re-enable "positioning tool" (the tool have the focus but I click another time on it) and it unlock the situation too 
and same for others tools ....

tested with my "1cm cube" -> same issue
[1cmCube.zip](https://github.com/Ultimaker/Cura/files/4645593/1cmCube.zip)

Same issue on 4.2.1 (_but the re-enable on positioning dosn't work I have to use another tool_ )
Same issue on 3.5.1
Same issue on 2.7.0

In my point of view this issue is linked with the initialization state of the object ... Cura don't "see" the object when you click on it the 1st time.. you have to "click" on a button to be able to "refresh" the status and enable the delete key shortcut

Note : I use only direct delete on my Imac "small" keyboard : [magic-keyboard-français](https://www.apple.com/fr/shop/product/MLA22F/A/magic-keyboard-fran%C3%A7ais) 

But if I use Fn+Delete the object is directly deleted event if it's a fresh stat of Cura
If I add another object, it's still not possible to delete it using the key, only Fn+delete work ...
As before, if I click on a tool (move, scale, mirror, .. , or slice) then this unlock the delete key feature

@Ghostkeeper  Please sorry to boring you again, I think the "issue" is a very old initialization issue and I'm pretty sure it could be easily fixed

Ghostkeeper, 2020-05-20:
Don't worry about boring me. I'm passionate enough about this application.

There's been some internal struggle with this. We are facing the problem that we don't have a development environment on MacOS any more since January. Only testing computers. We've been able to handle that so far by close communication, trying modifications on a tester's computer and reading the log output there. And in the meanwhile we've been asking IT for a new MacOS laptop to share with the team. But both of these methods have sort of been thwarted by everyone working from home due to COVID-19.

We've been struggling with this issue [before](https://github.com/Ultimaker/Cura/issues/6475), which led to a [change](https://github.com/Ultimaker/Cura/commit/bec513aa8a8c408cc7c9301659d7e846b581d3eb) in the 4.5 release. Apparently that doesn't work for you though.

To me this is not something I can easily fix. Since it's the act of opening certain dialogues that apparently breaks this, I think we need to debug whether those dialogues maybe still catch certain key events while they are invisible but still exist.

Ghostkeeper, 2020-05-20:
By the way, there's been some talk about this upcoming "64Florian" figure among the developers. You should know that we see that you're being active and helpful in debugging these issues, and it's really appreciated!

64Florian, 2020-05-20:
I'm very happy to know I'm a bit helpful ! Thank a lot for your message !
I try to do my best to help users and dev team (In my point of view, the reproducibility of a bug is 80% of the work for the correction ) 
I have a pretty great skill as soft analysts but not so great on pure coding ... 
I'll like to help more, but I'm a bit sort with git (in progress) and I try to setup the workspace but I was not able to do it :-( (I use a old MacOS Sierra) -> may be misunderstood something too...
Can you setup a remote debug ?
I'll try to use your tool in the ticket #7626 to see if I can find something interesting ...
I'll let you know.

Ghostkeeper, 2020-05-25:
Hardly. I've never tried to set up Cura's development on MacOS, I'm afraid. Only on Windows and Ubuntu.

It may make things easier if you set up an environment for just the Python front-end and pick CuraEngine, libSavitar and libArcus from our builds.

jhihn, 2020-06-30:
I guess I'm ahead of the curve. I filed 5198 ;-)

Anyway delete has never worked from at least 4.1 to.the current 4.6 on OSX Sierra, high Sierra or Catalina.

Ghostkeeper, 2020-07-02:
It's a bit of a mystery to me, this one. We're listening to the Delete key (keycode 8), and the Backspace key (46) both. Experiments with a MacOS laptop confirm that we're listening to the correct keycodes. We've tested with an external Apple keyboard on that laptop, and an external conventional ISO keyboard. It works for us there, but we still get reports of people that it doesn't work for them.

If you press that key in https://keycode.info/ which keycode does it give for you?

jhihn, 2020-07-02:
8

jhihn, 2020-07-02:
8

jhihn, 2020-07-02:
8

jhihn, 2020-07-02:
For what it's worth I don't think it's necessarily a key code issue. There could be an issue with capturing it, or identifying the model that is to be deleted. I would suspect that something is getting the keystroke that does not expected to.

64Florian, 2020-07-02:
> There could be an issue with capturing it, or identifying the model that is to be deleted.

I want to say the same ..
Because when back to cura after a long time (or on a fresh start) 
My model still visually selected but backspace (-> 8) doesn't work ..
If I try to use "sup" (using fn + backspace -> 49) the model disappear ...
If I want to use only backspace (8) I have to change the "tools" then press backspace (8)

fieldOfView, 2020-07-03:
> If I want to use only backspace (8) I have to change the "tools" then press backspace (8)

Is there a cursor (blinking vertical line) in any of the input fields of the open tool panel?

64Florian, 2020-07-05:
Hello @fieldOfView 
For: 
>  any of the input fields of the open tool panel?

No see any blinking vertical line cursor ... 
Please have a look on my video : 
http://floriantmonsite.free.fr/Cura/BackspaceCura.mov
I have added a "visual" keyboard to help 'to see' pressed keys (And I très to use the virtual keyborad with my mouse too ..

Ghostkeeper, 2020-07-06:
You're right. Since it works after changing away from the move tool, it's probably not a key code issue. Something in that tool could be capturing the key press.

Maybe it has something to do with this?

https://github.com/Ultimaker/Uranium/blob/9eaf0516cb8b3ce8c3b6e7e8e84386570e20667d/plugins/Tools/TranslateTool/TranslateTool.qml#L46

That should only grab focus when you press the tab key while the focus is already on the text field, though.

Still can't reproduce the issue here in Linux.

64Florian, 2020-07-06:
more or less "Just for curiosity", I try different way to enable the backspace key ...

Using "Command + A" or direct selection -> not working / Click on play pause plugin, then the backspace work : 
http://floriantmonsite.free.fr/Cura/usingSomethingElse1.mov (23 MB)

Using drop list (printer -> not working / using profile drop list, then the backspace work : 
http://floriantmonsite.free.fr/Cura/usingSomethingElse2.mov (27 MB)

Can't reproduce on windows ... :-( so only mac issue ?? (_my version is 10.12.6_)

@jbryson3  can you share you mac OS version please ?

Ghostkeeper, 2020-07-08:
Clicking the profile drop-down and starting a slice clears focus, so there is nothing in the move tool any more that has focus to catch any backspace key events. This focus clear is necessary to "apply" any settings that the user was in the process of typing a value for. The new value then gets used for the slice, or taken along to the profile you're switching to.

Maybe it's an idea to clear focus whenever a tool is activated? It's a bit weird to add code-wise, since opening a tool should have nothing to do with focus. But it would probably fix this, if we've theorised correctly.

Is any of you able to make changes to the QML files inside your application? I may have a solution, but since none of the developers can reproduce the issue we're having difficulty testing whether it works. The solution is to add this line:

    forceActiveFocus();

to `resources/qml/Toolbar.qml` at line 81. That is, at the bottom of the `onCheckedChanged:` event, just under the closing bracket of the `if (checked)` statement. It doesn't seem to have any negative consequences otherwise, adding this, but it could behave funky with some plug-ins maybe.

64Florian, 2020-07-09:
Hello @Ghostkeeper,
I have tested OK your changes 
You can have a look on my tests cases on this "big" video:
http://floriantmonsite.free.fr/Cura/L80-forceActiveFocus.mov (118 MB)

I can not be sure of any kind of border effect but as firsts shoot "it works great" !

I'll keep this modification on my version (_and try to keep in mind y 4.6.1 isn't anymore "clean"_ )

jhihn, 2020-07-10:
I'm not using many plugins, I think the only one I use is the octoprint plugin.

Gipielle, 2020-07-11:
The problem is present also with Cura 4.6.1 and catalina 10.15.5

jhihn, 2020-07-11:
@Gipielle Yes, this should be present everywhere, though it may be a platform (win/mac) lower level issue. I am running those versions. But the fix above worked for me

Ghostkeeper, 2020-07-13:
Thanks for testing, Jhihn! That was a lucky guess!
I'll add that fix there and hopefully that fixes the problem with the next release.

commit c38ee8a3 (2020-07-13): Clear focus when changing tools

This is a fix for a bug on MacOS where the delete key wasn't working. A text field from the tools was in focus, which caught the key event, preventing objects from being deleted.

Fixes #7754 and CURA-7585.
closed 2020-07-13 by Ghostkeeper