Ultimaker/Cura #7756: Adaptive layer reversed logic on curve surfaces on Cura 3.61, Win 7

Praetorianul, 2020-05-14:
[image]

Praetorianul, 2020-05-14:
I would expect to have a lower layer height on the most closed to horizontal portion, in the middle.

Praetorianul, 2020-05-14:
I just understood why is behaving like this. Because my variation step is very low but I need the small height layer suddenly. Would not be better that layer height variation to start earlier in order to reach the desired height at the right point? As it is now, the layer is starting getting lower after the point where is necessary. My variation step of 0.001 was meant to make change height less noticeable.

Ghostkeeper, 2020-05-15:
Yes, it's the limited layer height variation between one layer and the next, in order to prevent a sudden jump in flow rate.

Implementing a "lookahead" for adaptive layers is requested here: https://github.com/Ultimaker/Cura/issues/6641 so I'll close this as a duplicate then.

The implementation of that is quite involved, since CuraEngine records the slope of the surfaces that a layer crosses and decides on the height of the next layer based on that. If, based on a sudden change in slope, you'd have to go back a few layers to adjust gradually, you would need to re-evaluate the slope of the surfaces for the new cross sections. And then the slopes could be different again, warranting different behaviour a few layers back. Essentially we're looking at a global relaxation algorithm here if we are to implement this correctly.

closed 2020-05-15 by Ghostkeeper

Praetorianul, 2020-05-15:
Sounds fascinating what you are saying🙂. I was frustrated that I find so limited informations on builders groups and I found you. Was so obvious to check with you but I didn't expected so detailed and clarifying informations.

Adrian Stefanescu 
tel. +44 730 5553 556

Praetorianul, 2020-05-17:
[image]

Praetorianul, 2020-05-17:
I don't understand why is doing a higher layer where the angle is closest to 90 degrees. Should be reversed.