Ultimaker/Cura #7759: Cura 4.6.1 (and before) - Numbering of extruders confusing (0 and 1 or 1 and 2)

upatamby, 2020-05-14:
**Application version**
Cura 4.6.1

**Platform**
Windows 10

**Printer**
Ultimaker 3 Extended

**Reproduction steps**
1. Make some settings, e.g. set Build Plate Adhesion Extruder to Extruder 2 and set Brim width to a non default value
2. Click on the profile field to change it and look at the warning window

**Screenshot(s)**
[image]

**Actual results**
You see that the extruder numbering is weird: the Build Plate Adhesion Extruder is changed from 0 to 1 while the Brim Width applies to Extruder 2.

**Expected results**
Consistent numbering of the extruders: either 0 and 1 or 1 and 2, through the entire application.

Ghostkeeper, 2020-05-18:
Indeed, it should be "1" and "2". The bug is that this dialogue displays the internal representation of the setting values, for instance the drop-down options (like Build Plate Adhesion Type) also show our internal representation rather than the normal one (which is in most cases the same except capitalised).

upatamby, 2020-10-18:
There are more strange things: in the example below, Extruder 1 is referred to as Extruder 2
Isn't it time for a thorough clean up?
[image]