Ultimaker/Cura #7760: [4.5.0] Brim in holes with islands

AndreKR, 2020-05-14:
**Application version**
4.5.0

**Platform**
Windows 7

**Printer**
Anycubic i3 mega

**Reproduction steps/Screenshots**

The model:

[image]

Brim settings:

[image]

First layer:

[image]

**Actual results**
The left (on the screenshot) hole (the one with the island inside) has a brim, the right hole does not.

**Expected results**
This might be a misunderstanding on my part, but is it supposed to print a brim in the left hole with "Brim Only on Outside" selected?

AndreKR, 2020-05-14:
By the way, a Skirt with Skirt Distance = 0mm seems to give the expected result, despite the Skirt Distance input box being orange.

Ghostkeeper, 2020-05-18:
This is currently the expected behaviour, not a bug. The problem arises from how the brim lines are expected to join together like this:
[image]

We currently can't make the brim be interrupted by anything. It's always full contours. That would need to change.

fvrmr, 2021-07-01:
Closing this issue since it is a duplicate of https://github.com/Ultimaker/Cura/issues/4222 and we have a ticket on our backlog for this and is related to that issue, so we keep everything in one issue.

closed 2021-07-01 by fvrmr