Ultimaker/Cura #7762: (4.6.1) Initial Layer Horizontal Expansion not applied to inner edge

JohnYSOKC02, 2020-05-15:
**4.6.1**

**Creality Ender 5 stock hardware and  #firmware**

I'm printing a square tube with 4mm thick walls using a -0.2mm value to reduce the elephant's foot effect.
I noticed it works on all 4 outside edges but the value was not applied to the 4 inside edges.
Please confirm if this option is only meant for outside edges only. 
Will you further refine the programming to allow the value to be applied to the inside edges as well? 
Thank you. 

John.

smartavionics, 2020-05-15:
Hello @JohnYSOKC02 , please provide the Cura project file you used. Do File -> Save and then either zip up or rename the .3mf file and attach to this issue. Thanks.

JohnYSOKC02, 2020-05-15:
Will do.
Thx.

JohnYSOKC02, 2020-05-18:
Hi Mark Burton,

Please find the file in question.
Thanks.

John.
[CE5_Cube.zip](https://github.com/Ultimaker/Cura/files/4641638/CE5_Cube.zip)
[image]
[image]

smartavionics, 2020-05-18:
Thanks for the file. So the issue here is that the model is solid, it doesn't contain any holes. The fact that you have chosen 0% infill density (hence making the inside empty) does not change Cura's conviction that the model is soild. You need to embody the hole into the model for Cura to recognize that there is an inside edge there. Hope this helps.

Ghostkeeper, 2020-05-18:
Agreed!

If you're not able to do actual modelling, you can also try to reduce the Initial Layer Line Width a bit.

I'll close this issue, since it's working as expected.

closed 2020-05-18 by Ghostkeeper