Ultimaker/Cura #7764: [4.6.1] Ultimaker ABS update request stuck

imodre, 2020-05-15:
**Application version**
4.6.1

**Platform**
Win 10

**Printer**
Custom (FlyingBear Ghost 4s with MKS Wifi plugin from marketplace)

**Reproduction steps**
1. Update from 4.4.0 to 4.6.1
2. See that there is an update request in marketplace.
3. Update it
4. See that it's still there.

**Screenshot(s)**
[image]
[image]
[image]

**Actual results**
Unremovable material update request + account sync request.

**Expected results**
No such behaviour after trying to update this material.

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4634470/cura.log)
Searching 'ABS' in file could help maybe.

**Additional information**
Tried to run as administrator, no success.

mahtDFR, 2020-05-15:
Thanks for the report. I'm able to reproduce it. We're investigating

Phenix2984, 2020-05-17:
I think I have the same problem: from Cura 4.6.1 install, Cura ask for "Ultimaker ABS" plugin update at each start.
I don't use this plugin and I can't uninstall.

loulou-uoluol, 2020-05-18:
Hello, 
can't update also on Mac OS 10.15.4

mahtDFR, 2020-05-18:
This package was updated recently after a request from our product managers.
The version number is now higher than what ships with Cura, and the current Cura release has a bug that skips built-in packages when comparing.

There is a fix for this already in Cura, but it is yet to be released. Sorry for the hassle.

mahtDFR, 2020-05-18:
Relevant internal ticket(s):
CURA-6269
CURA-7366

Ghostkeeper, 2020-05-18:
The ticket where this fix was made was reported here: https://github.com/Ultimaker/Cura/issues/7426 . This is a duplicate.

closed 2020-05-18 by Ghostkeeper