Ultimaker/Cura #7765: [master] infill mesh hides selected object "glow"

smartavionics, 2020-05-15:
**Application version**
master

**Platform**
Linux with Qt 5.13.2

**Printer**
N/A

**Reproduction steps**

Loaded someone's project that has a large infill mesh that smothers most of the other models.

Selected one of the normal models that lie underneath the infill mesh.

View in prepare mode.

**Screenshot(s)**

[image]

**Actual results**

The object selection "glow" is not visible through the infill mesh.

**Expected results**

The highlighted outline of the selected object should still be visible through the infill mesh.

**Project file**

**Log file**

**Additional information**

Not really critical but a bit confusing inasmuch that you can select a model but it's not actually highlighted when viewed through the infill mesh.

Ghostkeeper, 2020-05-18:
Indeed. I guess that's because there is no exception for translucent objects, compared to the normal case where the objects that block the view are actually solid:

[image]

fieldOfView, 2020-05-18:
Wouldn't it be useful/better to have the selection outline show the full outline of the selected model(s) _regardless_ of what is in front of it?

nallath, 2020-05-18:
Wouldn't that look really weird if you have multiple objects selected?

Ghostkeeper, 2020-05-19:
I think it's either this, or as fieldOfView says everywhere.

Everywhere would be fine. It will make prints like this look a lot different:
[image]
However in a way it would make it more clear as well.

To implement that, the best way is to re-render the entire scene in the selection buffer rather than to have that as an additional pass, and then don't render objects that are not selected.

However we decided that this doesn't get enough priority for us over other things. We probably won't get around to implement this any time soon.

fieldOfView, 2020-05-19:
But if "someone" implements that, it will probably be merged?

Ghostkeeper, 2020-05-20:
I'd be in for it. What would you say, @Yian-Lai ?

Yian-Lai, 2020-05-25:
Yes, I agree. It will make it more clear.

closed 2023-06-24 by github-actions[bot]