Ultimaker/Cura #7766: [4.6.1] Dark theme bugs

dsvi, 2020-05-15:
`Ultimaker Dark` theme has a lot of issues.
For example if you go to `Profile` -> `Manage Profiles`, most of the text is not readable, since font is of black color on the black background.
[image]

**Application version**
4.6.1

**Platform**
Suse Linux KDE

mituritsyn, 2020-05-15:
Same issue
**Application version**
4.6.1
**Platform**
Kubuntu 18.04

Ghostkeeper, 2020-05-19:
Duplicate of https://github.com/Ultimaker/Cura/issues/4826 .

The elements in this preferences screen are partially system-themed, so if you change your system theme to a dark theme but not Cura, or if you change Cura to a dark theme but not your system, it's looking weird to the point of text being illegible.

closed 2020-05-19 by Ghostkeeper

dsvi, 2020-05-19:
Nah, that's not my case. I do have system-wide dark theme. And Cura is the only app affected by thins on my desktop.