Ultimaker/Cura #7767: Concentric Layer 1 Outside to Inside

davidlip-cura, 2020-05-15:
Cura currently lays down a 1st layer when set for Concentric from the inside to the outside.  The problem with this is there is nothing for that tiny little dot at the very center to stick to and it ends up sliding around with the nozzle making a mess of the first layer.  Seems like you would have a very nice anchor in the inner wall to work from when printing from outside to inside.

Tried Simplify 3D and it prints from the outside to the inside.  Makes for perfect concentric 1st layers.  No blobs.  No pushing the first layer around.  I was struggling to print a PETG 30x30 test box with Cura.  Simplify 3D printed a perfect box with no adjustments required.

Lines work but don't give the same appearance.

Researching I found other people with this problem but no solution.

Thanks,
David

Ghostkeeper, 2020-05-19:
From our testing, when printing from outside to inside the blobbiness tends to ruin your adhesion.

While you may have gotten it to work with another slicer, that slicer will use lots of different settings and techniques anyway. You can try printing with Cura 3.2, which had concentric patterns printed from outside to inside.

We decided to not make it a setting then, because the use case is very slim. We didn't find any use for it ourselves, and although it might be different for different printers in theory, it would even then be only relevant for approximately circular objects.

See also https://github.com/Ultimaker/Cura/issues/5033 of which this is a duplicate.

closed 2020-05-19 by Ghostkeeper

matthewatkins1, 2023-06-20:
Hi Team, 

I also need this as a setting, even if the use case is small, this one setting is stopping me from using Cura versions later than 3.xx.

also the concentric pattern now goes in both directions. Only cura 4.13 is able to achieve what I need