Ultimaker/Cura #7770: [4.6.1] Adaptive Layer Height not honoring Layer Height as minimum

tlhintoq, 2020-05-16:
**Application version**
Version 4.6.1 Windows

**Platform**
Windows

**Printer**
Printer is a custom created (not from template) Ender3pro (because I don't want that Creaweful mod in it) 

[image]

[image]

Asterchades, 2020-05-16:
0.2 with a variation of 0.12 will give you 0.08 to 0.32.

I'll be honest - I thought that was how it always functioned. "Variation" doesn't imply a direction, and nothing mentions a minimum.

tlhintoq, 2020-05-17:
> 0.2 with a variation of 0.12 will give you 0.08 to 0.32.
> 
> I'll be honest - I thought that was how it always functioned. "Variation" doesn't imply a direction, and nothing mentions a minimum.

Maybe I'm wrong then.  We all have that occasional "I could have sworn..." moments.  Maybe its something I always acheived through the use of the old parameters that went with it that have been replaced by "top size" and topography size".

Still... Have a minimum size _should be_ possible.  What if we don't want layers any thinner than .08 or .12 for example?  When I am doing a large item like cosplay armor on a 1mm nozzle, I don't need .04 layers.  .12-.44 does me just fine and doesn't add 3 more days of printing.

Asterchades, 2020-05-17:
Well, I mean, yeah... it should. But it already is, in so much as you set the mid-point and the variance which in turn controls both the minimum and maximum. It's perhaps not the most straight-forward way to implement it (if it were, you wouldn't have made this report) but the functionality is all there.

That range you've given, for example, would be a 0.28mm layer height with a variance of 0.16mm. Basically the layer height is the average of the range ((0.12 + 0.44) / 2) and the variance is the difference between the average and either end (0.44 - (0.12 + 0.44) / 2).

Trying to alter it would create two opposing, equally valid points: are you specifying your minimum desired layer height, or your maximum desired layer height, when choosing the base layer height? It makes sense both ways, and adding options for both in Experimental feels superfluous (even if it could work).

nallath, 2020-05-18:
> 0.2 with a variation of 0.12 will give you 0.08 to 0.32.
> 
> I'll be honest - I thought that was how it always functioned. "Variation" doesn't imply a direction, and nothing mentions a minimum.

Correct. This has always been the case. 

There is actually a pretty good reason for it. Profiles tend to be very much optimized for a given layer height. So although you might want to deviate from them a bit, deviating too much from them will hurt the results. So selecting the profile that is in "the middle" of the range is the best.

tlhintoq, 2020-05-18:
Ok.  My mistake.  Like I said I guess it was one of those "I could have sworn" moment where it turns out not to be.

Go ahead and close this bug

closed 2020-05-18 by tlhintoq