Ultimaker/Cura #7773: Loading PROJECT clobbers printer settings

THX2112, 2020-05-16:
Settings made to the printer (specifically start gcode) gets erased if a project is loaded. Original settings are not restored if a new project or Cura session is started.

**Application version**
461

**Platform**
Windows

**Printer**
Creality Ender 3 Pro

**Reproduction steps**
1. Add start gcode to printer settings (from Machine Settings).
2. Load project with different printer settings.
3. Restart Cura or create a new project.

**Actual results**
Gcode added to startup in step one is missing.

**Expected results**
Original start gcode from set one loaded.

fieldOfView, 2020-05-18:
Creating a new project never changes any settings back to a default.

Loading a project includes the full configuration of the printer, including all settings. Start gcode is a setting, so after loading a project the start gcode will be that of the project.

Ghostkeeper, 2020-05-19:
You can choose to not remove old printer settings in the pop-up that gets shown when you load a project file.
[image]
In the drop down in the top right, select "Create new". It will then create a new printer alongside your current one. It won't combine the settings, but you can switch.

closed 2020-05-19 by Ghostkeeper