Ultimaker/Cura #7774: Skirt starts with the inside towards outside! it was working great before when it was reversed.

ylectron, 2020-05-16:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

smartavionics, 2020-05-16:
This regression has been fixed for the next release.

See https://github.com/Ultimaker/CuraEngine/pull/1231

Ghostkeeper, 2020-05-19:
Duplicate of https://github.com/Ultimaker/Cura/issues/7747.

closed 2020-05-19 by Ghostkeeper