Ultimaker/Cura #7776: Processing circles is done using many breaks and "sticky work"

FPrefect20, 2020-05-17:
**Application version**
Cura 4.6.1

**Platform**
Windows 10 (64 Bit)

**Printer**
Anycubic I3 Mega S

**Reproduction steps**
1. Designing a plain circle (e.g. 0.1mm in height)
2. Processing with cura using "concentric pattern"

**Screenshot(s)**
screenshots of processed standard-circle
[image]
[image]

**Actual results**
The 1st layer of the circle is printed in a chaotic way with many direction changes and jumpes. 
There is no continuous circle line and so the printing result is full of pimples.
This is even a bigger problem with critical filament types. 

**Expected results**
Concentric, continuously printed circle line.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

smartavionics, 2020-05-17:
Hi, please provide a project file that shows this problem. Do File -> Save and then either rename or zip the .3mf file and attach to this issue. Thanks.

FPrefect20, 2020-05-18:
Hi, 
the issue can be observed with the enclosed simple "almost circle".
(almost circle because TINKERCAD divides the circle in 64 segments)

As some point CURA is leaving the concentric pattern and doing some "lining" - 
irrespective of the pattern settings which causes even a longer printing time.

[Issue7776.zip](https://github.com/Ultimaker/Cura/files/4644980/Issue7776.zip)

Example layer no. 5:
[image]

Ghostkeeper, 2020-05-19:
Right, this is a duplicate of https://github.com/Ultimaker/Cura/issues/5110 then.

Your first and second screenshots also show a little jaggy line which has been fixed in the upcoming version. The rest of the issue in the middle of the curve seems to be the result of the shape not being of uniform width, perhaps due to staggered vertices on the inner curve vs. the outer curve.

closed 2020-05-19 by Ghostkeeper