Ultimaker/Cura #7777: Wall Thickness Based on Line Count

Tuz550, 2020-05-17:
Is it possible to make the wall thickness variable dependant on the wall line count variable? I'm currently using a specific line width accurate to two decimal places and would find it significantly easier to change the wall line count setting than the wall thickness setting when I want to change the number of walls. (eg. if I'm using a line width of 0.42mm, it would be easier to change the wall line count to 3 and have it calculate a wall thickness of 1.26mm instead of reaching for a calculator).

fieldOfView, 2020-05-17:
If you change the Wall Line Count, the Wall Thickness is ignored (and the field Wall Thickness will be disabled because the value is unused).

64Florian, 2020-05-17:
as said by @fieldOfView you can enable this feature and have the flowing display
[image]

Ghostkeeper, 2020-05-19:
Perhaps easier, you can also right-click on the Wall Thickness setting and then click "Hide this setting".

But yes, if the Wall Line Count is customised, it overrides the Wall Thickness. I hope that solves your problem.

closed 2020-05-19 by Ghostkeeper