Ultimaker/Cura #7779: [4.6.1] Cura White Screen on Startup, not responding

Brzam, 2020-05-17:
**Application version**
4.6.1

**Platform**
Windows 10 (GFX = AMD Radeon RX 5700 XT)

**Printer**
AnyCubic i3, but issue occurs prior to this

**Reproduction steps**
Unsure of the steps, this bug appears to occur randomly, but frequently for me.
**Screenshot(s)**
[image]

**Actual results**
Cura displays a white screen with no option to continue, the only option being to force close the application. This issue disappears when deleting the entire contents of %appdata%/cura, but obviously all settings are deleted.

**Expected results**
Cura starts up normally.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
[stderr.log](https://github.com/Ultimaker/Cura/files/4640381/stderr.log)

**Additional information**
This issue has been occurring for me since 4.5 but have has been heappening more frequently recently.

nallath, 2020-05-18:
Do you have extra /multiple montitor(s) attached?

Brzam, 2020-05-18:
I do indeed, am running 3 1920x1200 monitors.

fieldOfView, 2020-05-18:
Try disabling two of them temporarily. If Cura starts succesfuly with a single monitor, go to the General preferences and uncheck the "Restore window position on start". Then you can reenable the other screens, and Cura should always open on your primary screen.

Brzam, 2020-05-18:
Done! Will re-open this if the issue persists, thank you both :)

closed 2020-05-18 by Brzam

d-rez, 2024-03-22:
This is a bug still in 5.4.0 and 5.7.0beta and it needs to be fixed. Disconnecting / disabling monitors isn't a solution...

sajjadkurd1999-lang, 2026-09-19:
Pubg white body