Ultimaker/Cura #7781: 4.6.1 - Tree supports not rendering properly

batesman1, 2020-05-17:
**Application version**
4.6.1

**Platform**
Windows 10

**Printer**
Ender 3 Pro

**Reproduction steps**
Set up custom supports. Disabled supports. Enabled tree supports. Tree supports render with lines and not tree supports. Raft enabled.

**Screenshot(s)**
[image]

**Actual results**
Line support are drawn instead of tree supports.

**Expected results**
Tree supports should be generated instead of line supports.

**Project file**
[CE3PRO_Half Ore Death Cleric Paladin.zip](https://github.com/Ultimaker/Cura/files/4640768/CE3PRO_Half.Ore.Death.Cleric.Paladin.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4640770/cura.log)

batesman1, 2020-05-17:
I don't know if this is expected behavior or not, but this happens when support wall line count under supports is set to 0.

closed 2020-05-17 by batesman1

Ghostkeeper, 2020-05-19:
Indeed, tree supports alter the defaults for the support settings, but those support settings still work the same way. By default Tree Support sets the Support Wall Line Count to 1 and the infill density to 0%, but if you change that you get a result like this.

That is expected behaviour.