Ultimaker/Cura #7784: Hover over gradient to see value in preview

DkOuvE, 2020-05-18:
Right now it's barely possible to see what speed anything is going at, just that some parts are faster than others.

[image]

I have an idea of hovering over the gradient kind of like a colour picker to see the exact value:
[image]

mahtDFR, 2020-05-19:
We feel hovering over the gradient is easier to make but less useful. Hovering over the gradient alone wouldn't allow you to compare that value on the object easily. Or did you intend it to work with the object on the build plate as well?

I think we'd like to defer this for now, but if someone can make a pull request we will gladly review it.

Glodigit, 2023-07-01:
I still think this is a useful feature. 

A more advanced way would be similar to how Fusion 360 tackles this issue in its simulation view, where the limits are draggable. It looks quite similar to changing the limits of viewable layers in Cura, where values outside the range have a dark, transparent texture.