Ultimaker/Cura #7787: Incorrect brim flow

emil-e, 2020-05-18:
3

**Application version**
4.6.1

**Platform**
Windows 10

**Printer**
Custom (BLV mgn cube)

**Reproduction steps**
1. Set nozzle size to 0.25mm
2. Set line width to 0.28mm
3. Set layer height to 0.06mm
4. Set initial layer height to 0.2mm.
5. Set initial layer line width to 200%.
6. Add a brim.

**Screenshot(s)**
[image]
[image]

**Actual results**
Brim is printed with the correct line spacing but with incorrect flow. This can be verified by exporting the GCode and then reimporting it into Cura.

**Expected results**
Brim has correct flow.

**Project file**
[CFFFP_-.zip](https://github.com/Ultimaker/Cura/files/4646683/CFFFP_-.zip)

**Log file**
Not relevant.

Ghostkeeper, 2020-05-19:
Cura's own g-code viewer seems to agree. Some brim lines are thinner than others, even though none of these keys have any per-object settings.
[image]

mahtDFR, 2020-05-20:
I could reproduce this too, here's a gcode output.

[4.6.1 CFFFP_- BLV Keycap PLA machine_nozzle_size 0.25mm line_width 0.28mm layer_height 0.06mm layer_height_0 0.2mm initial_layer_line_width_factor 200%.gcode.zip](https://github.com/Ultimaker/Cura/files/4656877/4.6.1.CFFFP_-.BLV.Keycap.PLA.machine_nozzle_size.0.25mm.line_width.0.28mm.layer_height.0.06mm.layer_height_0.0.2mm.initial_layer_line_width_factor.200.gcode.zip)

I can't seem to test the project file with 4.5 or lower, as I get this:
[image]

emil-e, 2020-05-20:
I recall seeing weirdness related to this even before increasing the initial layer line width. I had sliced one model which printed correctly and then sliced another one which had adhesion issues, I'm assuming this was the cause.

batesman1, 2020-05-22:
+1 more for this problem. All of my prints are failing on 4.6.1

batesman1, 2020-07-20:
Is anyone looking at this yet? I'm having to print everything on a raft now...

smartavionics, 2020-07-20:
Could it be that the initial_layer_line_width_factor term needs to be added to the skirt/brim config initialisation? i.e. in here...
```
        skirt_brim_config_per_extruder.emplace_back(
                PrintFeatureType::SkirtBrim
                , train.settings.get<coord_t>("skirt_brim_line_width")
                    * ((mesh_group_settings.get<EPlatformAdhesion>("adhesion_type") == EPlatformAdhesion::RAFT) ? 1.0_r : line_width_factor_per_extruder[extruder_nr]) // cause it's also used for the draft/ooze shield
                , layer_thickness
                , train.settings.get<Ratio>("skirt_brim_material_flow") * ((layer_nr == 0) ? train.settings.get<Ratio>("material_flow_layer_0") : Ratio(1.0))
                , GCodePathConfig::SpeedDerivatives{train.settings.get<Velocity>("skirt_brim_speed"), train.settings.get<Acceleration>("acceleration_skirt_brim"), train.settings.get<Velocity>("jerk_skirt_brim")}
            );
```

However, that could cause another problem as this config is also used for the draft shield (so it says) which should not have the line width modified for layers above the first.

smartavionics, 2020-07-20:
Ignoring the draft shield issue for a moment, the above suggested change makes the code look like..
```
        skirt_brim_config_per_extruder.emplace_back(
                PrintFeatureType::SkirtBrim
                , train.settings.get<coord_t>("skirt_brim_line_width")
                    * train.settings.get<Ratio>("initial_layer_line_width_factor")
                    * ((mesh_group_settings.get<EPlatformAdhesion>("adhesion_type") == EPlatformAdhesion::RAFT) ? 1.0_r : line_width_factor_per_extruder[extruder_nr]) // cause it's also used for the draft/ooze shield
                , layer_thickness
                , train.settings.get<Ratio>("skirt_brim_material_flow") * ((layer_nr == 0) ? train.settings.get<Ratio>("material_flow_layer_0") : Ratio(1.0))
                , GCodePathConfig::SpeedDerivatives{train.settings.get<Velocity>("skirt_brim_speed"), train.settings.get<Acceleration>("acceleration_skirt_brim"), train.settings.get<Velocity>("jerk_skirt_brim")}
            );
```
And the resulting view of the first layer looks more like it has 200% line width...
[image]

smartavionics, 2020-07-20:
Not woken up yet so I've only just noticed that layer_nr is available so that change could be...
```
        skirt_brim_config_per_extruder.emplace_back(
                PrintFeatureType::SkirtBrim
                , train.settings.get<coord_t>("skirt_brim_line_width")
                    * ((layer_nr == 0) ? train.settings.get<Ratio>("initial_layer_line_width_factor") : 1.0_r)
                    * ((mesh_group_settings.get<EPlatformAdhesion>("adhesion_type") == EPlatformAdhesion::RAFT) ? 1.0_r : line_width_factor_per_extruder[extruder_nr]) // cause it's also used for the draft/ooze shield
                , layer_thickness
                , train.settings.get<Ratio>("skirt_brim_material_flow") * ((layer_nr == 0) ? train.settings.get<Ratio>("material_flow_layer_0") : Ratio(1.0))
                , GCodePathConfig::SpeedDerivatives{train.settings.get<Velocity>("skirt_brim_speed"), train.settings.get<Acceleration>("acceleration_skirt_brim"), train.settings.get<Velocity>("jerk_skirt_brim")}
            );
```

smartavionics, 2020-07-20:
Unfortunately, that change only seems to affect the layer view, the gcode is still wrong!

[image]

smartavionics, 2020-07-21:
Hi @emil-e , I made a Cura build (20200721) that has a fix for this issue if you wish to try it out. You can find my builds at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file there before using.

nallath, 2020-07-21:
I've just tested this with the pull request from @smartavionics and the problem seems to be fixed.

closed 2020-07-21 by nallath