Ultimaker/Cura #7788: When you have 2 printer, slicer only uses one.

karayakar, 2020-05-19:
I have monoprice Delta and MP mini printer from monoprice. 
After adding Delta printer, slicer only uses Delta settings (it shows MP mini) and MP mini can't print objects. 

[image]

;FLAVOR:Marlin
;TIME:13817
;Filament used: 6.554m
;Layer height: 0.13125
;MINX:-40.075
;MINY:-40.075
;MINZ:0.263
;MAXX:40.075
;MAXY:40.075
;MAXZ:19.913
;Generated with Cura_SteamEngine 4.6.1
M140 S70
M105
M190 S70
M104 S235
M105
M109 S235
M82 ;absolute extrusion mode
G21;(metric values)
G90;(absolute positioning)
M82;(set extruder to absolute mode)
M107;(start with the fan off)
G28;(Home the printer)
G92 E0;(Reset the extruder to 0)
G0 Z5 E5 F500;(Move up and prime the nozzle)
G0 X-1 Z0;(Move outside the printable area)
G1 Y60 E8 F500;(Draw a priming/wiping line to the rear)
G1 X-1;(Move a little closer to the print area)
G1 Y10 E16 F500;(draw more priming/wiping)
G1 E15 F250;(Small retract)
G92 E0;(Zero the extruder)
G92 E0
G92 E0
G1 F2400 E-4.5
;LAYER_COUNT:151
;LAYER:0
M107
G0 F4320 X-6.032 Y-37.99 Z0.263
;TYPE:SKIRT
G1 F2400 E0
G1 F1800 X-5.637 Y-38.465 E0.02364
G1 X-5.172 Y-38.9 E0.04801
G1 X-4.655 Y-39.273 E0.07241
G1 X-4.097 Y-39.579 E0.09676
G1 X-3.504 Y-39.812 E0.12115
G1 X-2.887 Y-39.969 E0.14551
G1 X-1.969 Y-40.057 E0.1808

.........

karayakar, 2020-05-19:
I have verified with Cura 3.3.1 and it doesn't do that. 

[image]

fieldOfView, 2020-05-19:
When creating an issue, please fill in the issue template.

I don't know what you mean by this:
> After adding Delta printer, slicer only uses Delta settings (it shows MP mini) and MP mini can't print objects.

Are you talking about which printer gets printed to when printing via USB?

mahtDFR, 2020-05-19:
Please fill in the issue template so we are able to help you:

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-05-20:
Looks like it somehow changed the Center Is Origin machine setting for the Monoprice printer?

karayakar, 2020-05-30:
I was able to manage uninstall and install. Now it is working fine.

closed 2020-06-02 by nallath