Ultimaker/Cura #7789: [4.6.1] continually prompts to Update ABS Material, and wont load Proto-Pasta materials from Marketplace

gleonard3, 2020-05-19:
**Platform**
(Linux AppImage)

**Printer**
(Ender 3)

**Reproduction steps**
1. (Observed Marketplace notification.  Signed in.  Located available Update for ABS.  Hit Update.  Restarted Cura, Marketplace notification persists), Retried, Retried)
2. (Attempted to install some Proto-pasta materials.  Hit Add, checked Manage Materials, Nothing shown under this manufacturer)

**Actual results**
(Marketplace notification persists. Additional materails not found)

**Expected results**
(ABS should be updated.  No notification should appear, new materials should be available in Manage Materials UI.

[cura_bug.log](https://github.com/Ultimaker/Cura/files/4648192/cura_bug.log)

mahtDFR, 2020-05-19:
1. Is reported and fixed already. See discussion here: https://github.com/Ultimaker/Cura/issues/7426
2. Can you verify how you added the Proto Pasta materials? from the web or within Cura?

gleonard3, 2020-05-19:
I tried both methods, thru the web and Syncing, or using the Marketplace from the Cura UI.  Neither of these we're successful.  Then I had the idea to check the ~/.local/share/cura/4.6/materials/ folder, and there appeared to be a Proto-Pasta folder there with a material file inside.  I attempted to manually import that, and still, no Proto-Pasta manufacturer in the Materials Manager.  I then opened this local Proto-Pasta material file in a Text Editor, and it appeared it was just a straight copy of Generic PLA.  At this point, I'm not sure it was a good idea to try that.  All this activity should be captured in the previous log I attached.  If you need something further, please let me know.

fieldOfView, 2020-05-19:
All the materials in the Marketplace only have profiles for printers that use 2.85mm material (ie: Ultimaker printers). Your printer has 1.75mm material. None of the materials you download from the Marketplace will show up for your printer.

gleonard3, 2020-05-20:
If that's the case, it would be good if that was made clear somewhere in the UI, so that people aren't led down this path.  That being said, I do have eSun PETG material settings installed, which I'm pretty sure I got thru this interface.  Should I not trust that those settings are compatible with my Ender 3?

fieldOfView, 2020-05-20:
eSUN PETG comes with Cura. You did not get it from the Marketplace.

nallath, 2020-05-20:
We're still working on adding a material compatibility list to the marketplace.

closed 2020-05-20 by nallath