Ultimaker/Cura #7792: [4.6.1] Skin Edge Support Layers missing layers

DkOuvE, 2020-05-19:
**Application version**
4.6.1

**Platform**
Windows 10

**Printer**
Custom Cartesian Profile

**Reproduction steps**
1. Support layers of 5 or less: All layers sliced correctly
2. Support layers of 6 or above: Missing skin edge supports

**Screenshot(s)**
[image]

**Actual results**
When support layers are set to the equivalent of 6 layers or more, there is a gap between where the support layers start and when they end. Usually a few support layers, then no support layers, then 1 support layer, then the skin.

**Project file**
[SpoolBack.zip](https://github.com/Ultimaker/Cura/files/4653513/SpoolBack.zip)

smartavionics, 2020-05-20:
Hi, thanks for the report and project file. I think I know what's happening here, I just need to come up with a fix.

smartavionics, 2020-05-20:
I have a fix which I am testing, will submit a PR when happy...

[image]

DkOuvE, 2020-05-20:
That's fast. Looks promising already.

Ghostkeeper, 2020-05-22:
Probably happens because it looks only a certain number of layers higher to see if there is skin there?

smartavionics, 2020-05-22:
No, it's because there are multiple skin layers that require the edge support (the skins are on different layers). I have a fix which I will submit as a PR soon.

smartavionics, 2020-05-25:
If you wish to try my latest build (20200525) which should be on dropbox in the next hour it has a fix for this issue.

You can find my builds at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0&lst=

Please read the README.md file there before using.

bastion-bastion, 2020-05-29:
I will try this to. My infill is a mess.

closed 2020-08-04 by rburema

rburema, 2020-08-04:
I just merged this to the main branch, so it's a lock-in for 4.8. We thought the change was a bit too large for the upcoming 4.7 (the beta is _very_ close), so if you want the official Ultimaker version (rather than the bleeding-edge builds of @smartavionics ) of this you'll have to wait a few months.

Glodigit, 2020-08-04:
Looks like I'll be backing up my configs and trying out those builds then, since it seems that all the cool features are going to be in the pipeline for a while.