Ultimaker/Cura #7796: custom printers default to dfp instead of g-code

calboy20, 2020-05-20:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
custom

**Reproduction steps**
1. created new custom printer
2. setup printer (anycubic mega zero) and import model save to SD card

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
saves print file as ufp file instead of G-code

**Expected results**
defaults to g-code or gives an option to change it.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-05-20:
Are you very sure you added a Custom FFF Printer, and not an Ultimaker S5?

A screenshot of your buildplate will confirm this.

calboy20, 2020-05-20:
[image: image.png]

fieldOfView, 2020-05-20:
Unfortunately, mail attachments don't show up in github issues. You have to add the image in the web interface, or post a link to the image.

Ghostkeeper, 2020-05-25:
Neither Anycubic nor Custom default to UFP for me.

A log file could help as well. Please fill in the rest of the template.

closed 2020-06-19 by no-response[bot]