Ultimaker/Cura #7797: Tree Support on different Models ALWAYS connect. Support Blocker is not working

pavo141, 2020-05-20:
**Application version**
4.6.1 The newest Version at the Moment

**Platform**
Mac OS. 

**Printer**
CR-10s

**Reproduction steps**
1. Change Support X/Y Distance
2. Tried Support Blocker

**Screenshot(s)**
[image]

[image]

**Actual results**
The support still connects... 

**Expected results**
I would love to have individual support for each model.

**Project file**

[CCR10S_Github_Issue.3mf.zip](https://github.com/Ultimaker/Cura/files/4659193/CCR10S_Github_Issue.3mf.zip)

64Florian, 2020-05-21:
Hello @pavo141
I manage to made this with 4.6.1 : 
[image]
I Have enable "Print sequence" in "One at Once" mode 
[image]
But I have to "tune" the printer settings (I made a "dangerous" modification in the settings ... so **Please, when you test this idea check your layer before and stay close your printer the first time to be sure you don't break anything with my "bad" idea !** )

The modification is on "Gantry Height" value in the printer settings (I put 200 mm instead of 25) 
=> You have to be careful about 1st and second item to be done to prevent breaking 1st print with the Gantry (nozzle bar support)
[image]

I have a other idea: 
If you tilt your models 45° X and 45°Y you "don't need" any support in fact and the print quality will be better (_look but not truly as strong, this is true. And it can be more expansive to add the tree support in sit case. but basic support may be interesting) 
[image]

pavo141, 2020-05-21:
Hey Florian,

Thanks for help. The Print at once option is really interesting but it won’t help me.

I was going to print a lot of the model and up to 8 for one print. It just bothering me that it takes even longer to print 8 at one Print-job than at 8 Print-Jobs. The reason is the connected tree support. It causes also a slightly increase in Material.

I was just hoping to get rid of it so I could save some time and Material. 
Flipping the Model and use other Support just results in more Material use.

Thanks very much for help! I appreciate your effort!

64Florian, 2020-05-21:
>  it takes even longer to print 8 at one Print-job than at 8 Print-Jobs

It should be really close with "Print sequence" in "One at Once", no ?
4h39 for 2 objects in "One at Once" and 21g
2h19 for one object and 11g
For 8 items, the nigth you "are not able" to remove the item and start a new job (Of cours you can tweak you GCode to do that by asking the printer to remove the object from build plate ... but it's something a bit risked for the printer / the model to be sure all is clean before stating a new object and can dommage the printed part or the printer ...

Another approach could be made the distance between each object closer as possible 
This is a test for 6 model in one row ... but of course due to the limitation it cost time and materials :-(
[image]

Another approach could be use "Mesh type: Modify settings for overlaps":
Disable global setting tree mode and enable it in the overlap (I use a stretched cube to apply the settings)
But unfortunately I didn't manage to have a result :-(
[image]

pavo141, 2020-05-21:
Hey,

Thank you very much. Placing all objects closer together would be a good solution!
But I would love to tweak the G Code because my prints are looking far better if I print only one than 8 at once. This is an awesome idea, it should been mine :D

I would love to see if the devs see this and maybe can implement a button where we can choose "Tree support per Model" so the Support at least doesn't connect.

64Florian, 2020-05-21:
> implement a button where we can choose "Tree support per Model"
Totally Agree with you !

> I would love to tweak the G Code

hmm ...
Manually you can ..
it's not easy but you can ...
You have to build 8 GCode of 8 different positions of your object then merge all "layer" one by one using a soft to manage extruder position ... It's really time consuming to crate this tool and if you have to made "only" 8 items it's only for fun regarding the amount of work needed by this task (but if you have more* it can be interesting to do) (the soft to do this merge will use  ";LAYER:XXX" as trigger and extruder position must be updated according the amount of material used by each model) 

One thing I have learned  by the past, If one item work well, you can have big trouble if you try to print 2 items at the same times ! (in fact the material lose his heat between each layer, the build plate change his temperature due to fan position, shape of the object and so more ...)
and if something goes wrong on one object, you may lose all items in one shoot :-(

The advantage of the "One at Once" it you keep your great support, if something goes wrong on one item you will not lose all your objects you can print at least 6 items by print and you can do this with the current version of Cura (but you have to be really careful for print order and on the 1st print)

more* : 24 at least ...

pavo141, 2020-05-21:
I was thinking about modifying the GCode to print one after one like the guy in this video:

https://youtu.be/avlengYsJdw

This would solve my problem by getting better results by printing on at a time and also save Material.

64Florian, 2020-05-21:
I already saw this video that's why I sayed befor
> GCode to do that by asking the printer to remove the object from build plate ... but it's something a bit risked for the printer / the model to be sure all is clean before stating a new object and can dommage the printed part or the printer ...

For 
> I was interested to tune my printer to a “24/7” Printer

In this case you don't need any custom tuning in Cura (Just add custom command and copy post in a giant file ;-) )
But according the size of your object and the actual brim, you must wait your build plate to cool down to 25°c before thinking to remove the product with you nozzle as shown in this video ;-)

Ghostkeeper, 2020-05-25:
Support Blocker, as you may know, doesn't actually block the support structure but prevents support from being generated at the overhangs.

With a bit of trickery you can create an actual "support blocker" though as you would expect it to behave. Instead of clicking on the Support Blocker Button you make it a normally printed object and give it 0 walls, 0 top/bottom thickness and 0 infill. Since Tree Support (and normal support) will not go through the models, this will block support. Even if that model is not being printed since it has no skin, walls or infill. You may need to move the object 1 layer into the air though, to prevent it from generating a brim as well.

[image]

> implement a button where we can choose "Tree support per Model"

Support in general is not specific to any model at the moment. Essentially CuraEngine generates one support object for the entire print, and you can adjust the settings of that object but not for the supports that happen to be underneath a certain model.

Changing that is problematic. Not just because it'd be a major refactor of basically the entire support generation stage (one of the 6 stages of the slicing process) but also because it makes supported dual extrusion models very weird.

pavo141, 2020-05-28:
Hey, 

thanks for the answer. This would actually solve my problem! But always if I choose the Support Block and Normal Mode it won't let me change Wall Thickness.

[image]

Any Suggestions?

Ghostkeeper, 2020-05-29:
That happens if you've customised the Wall Line Count setting. Wall Line Count overrides Wall Thickness. Set the Wall Line Count to 0 then.

CCS86, 2021-05-11:
Is this issue slated for improvement?

I'm not sure why this is labeled as a "Mac OS platform" issue. I experience this on Windows 10 x64.

The behavior to join support bases, when left unchecked, creates a very undesirable situation. Any time connecting support bases causes an increased amount of support time/material, it has defeated the purpose and become detrimental.

The behavior is very unpredictable too. Some supported areas seem to gravitate towards being isolated, and others are determined to connect to one another:

[image]

[SupportBase.3mf.txt](https://github.com/Ultimaker/Cura/files/6461533/SupportBase.3mf.txt)

Ghostkeeper, 2021-05-18:
The MacOS label was wrong. It's platform-independent. It's also not really "settings"...

> The behavior is very unpredictable too. Some supported areas seem to gravitate towards being isolated, and others are determined to connect to one another:

The behaviour is a heuristic, and your case is explainable if you know the heuristic. Basically what happens is that on every layer, Cura calculates the smallest set of (horizontal) connections between the tree branches (technical term: Minimum Spanning Tree). It then moves all branches towards the centre of all of its neighbouring branches. As a result, they tend to spread out in a line locally, and then the endpoints close in.

In your case, it likely determined that the tree with the green arrow in the middle is already in the centre between its neighbours (the two closest endpoints of the blobs of support with red arrows). The tree on the left with the green arrow would try to go towards its nearest neighbour on the right side of that object, but is blocked by the object itself in its path. Eventually it would make its way around the object unless it ends up in a local optimum.

I think an improvement to that heuristic to improve this behaviour would break up the MST into several pieces, depending on how far they are away from each other. That means that if the branches start off more than a certain distance from each other they would not converge unless new branches start lower that bridge the gap. I think the distance though should depend on the Z coordinate, because if the branches start off very high the user would likely expect them to group up together anyway. Otherwise you'd get more separate trees in bigger scale prints where they could easily come together.