Ultimaker/Cura #7798: [4.6.1] Multiple Extruders  Option Blocked  -  Creality w RepRap

magnokneeto, 2020-05-21:
Hi,
I'm trying to set up a profile to run multiple extruders on my Creality CR10 Mini that is upgraded with a Duet2 and RepRap firmware.  On the machine profile page it shows a dropdown for the number of extruders but the only option is "1" and this number cannot be overridden.  Am I missing something?

The Duet2 comes stock capable of handling 2 extruders.  Using Expansion boards this quantity can be greatly increased.

CR10 Mini
Duet2 w RepRap
Cura 4.6.1
Win 10

[image]

64Florian, 2020-05-21:
Hello @magnokneeto,
as dirty workaround you can create a "new" printer with "custom FFF"

[image]

I try to change the number of extruder on my Mac for a Tevo Tarantula too, and, event I try to change in cfg files, my settings are overloaded when I open Cura 
As I know, you could copy and rename proper cfg files and change "all" mandatory settings in each files, but it could made a big mess in the application, so, as workaround it still safer to manually fill all settings ...

I hope someone give you a better answer as me ...

magnokneeto, 2020-05-21:
OMG.  Thank you!
Your dirty work around is the freshest, cleanest thing i can do.  I'm just going to write my own headers and footers. 

You helped me to realize that i already adjust all the machine inputs for my printer build space that Cura uses to generate the virtual build space.  Really all these profiles do is pre-propogate those fields, a great service for people who don't know their printer yet, and script header and footer codes.  Unfortunately someone started scripting config overrides in them in 4.6.1 and not all these firmware use the same units...... Not sure why anyone would want to override their default settings to move their feeder at 500mm/s....in RepRap that setting works out to only 8.3mm/s. But I digress. 
Net result : winning

nallath, 2020-05-25:
Adding a custom amount of extruders is a pretty heavy modification to a machine. At that point, the machine isn't considered as a "standard" machine by us anymore. So there are indeed two options:
1. Create a custom fff printer
2. Create a new definition with the correct number of extruders.

closed 2020-05-25 by nallath