Ultimaker/Cura #7799: Scale X Y Z UI value/ boxes seem to be incorrectly wired or placed with respect to each other.

twobob, 2020-05-21:
**Application version**
4.6.1

**Platform**
Windows 10

**Printer**
Creality CR10 

**Reproduction steps**

1. I selected an object and chose SCALE
2. I dragged to tool handle in the Z axis to increase the value and observed the %age value being updated in another column.

**Screenshot(s)**
[image]

**Actual results**
The Y axis %age increased rather than the Z axis %age value

**Expected results**
The relevant axis %age changing relative to the manipulated intended measurement

**Additional information**
It results in wacky resizes when clicking around in the zoom boxes after resizing something absolutely.

ssavickiy, 2020-05-21:
I also encountered this error 
[image]

64Florian, 2020-05-21:
look like it's the old bug #4136

twobob, 2020-05-22:
Ah thanks. I did search trying to find it.

fieldOfView, 2020-05-22:
As long as the model is not rotated in Cura, the boxes will be "correctly wired".

twobob, 2020-05-22:
I see, so it's some kind of projection issue. Fine. Thus my hack should work for all cases. Good to know

Ghostkeeper, 2020-05-25:
Yeah, it's a well known but decently difficult bug.

Closing this as duplicate of https://github.com/Ultimaker/Cura/issues/4136, https://github.com/Ultimaker/Cura/issues/3625, https://github.com/Ultimaker/Cura/issues/4309, https://github.com/Ultimaker/Cura/issues/4386, https://github.com/Ultimaker/Cura/issues/5221, https://github.com/Ultimaker/Cura/issues/5189, https://github.com/Ultimaker/Cura/issues/5310, https://github.com/Ultimaker/Cura/issues/5462, https://github.com/Ultimaker/Cura/issues/5599, https://github.com/Ultimaker/Cura/issues/5602, https://github.com/Ultimaker/Cura/issues/5603, https://github.com/Ultimaker/Cura/issues/5872, https://github.com/Ultimaker/Cura/issues/5916, https://github.com/Ultimaker/Cura/issues/5965, https://github.com/Ultimaker/Cura/issues/6148, https://github.com/Ultimaker/Cura/issues/6150, https://github.com/Ultimaker/Cura/issues/6238, https://github.com/Ultimaker/Cura/issues/6370, https://github.com/Ultimaker/Cura/issues/6619, https://github.com/Ultimaker/Cura/issues/6770, https://github.com/Ultimaker/Cura/issues/6829, https://github.com/Ultimaker/Cura/issues/6890, https://github.com/Ultimaker/Cura/issues/6971, https://github.com/Ultimaker/Cura/issues/7102, https://github.com/Ultimaker/Cura/issues/7142, https://github.com/Ultimaker/Cura/issues/7237, https://github.com/Ultimaker/Cura/issues/7388, https://github.com/Ultimaker/Cura/issues/7446, https://github.com/Ultimaker/Cura/issues/7448, https://github.com/Ultimaker/Cura/issues/7582, https://github.com/Ultimaker/Cura/issues/7638, https://github.com/Ultimaker/Cura/issues/7703, https://github.com/Ultimaker/Cura/issues/7768 and https://github.com/Ultimaker/Cura/issues/7778. Yup, it's bad.

closed 2020-05-25 by Ghostkeeper