Ultimaker/Cura #7800: Brim not generated between tightly packed parts

viesturz, 2020-05-21:
**Application version**
4.6.1

**Platform**
win64

**Printer**
Custom

**Reproduction steps** 
1. Load https://www.thingiverse.com/download:4324151
2. Unzip and load compleate_buildplate...
2. Select 6m brim and slice. 

**Screenshot(s)**
https://imgur.com/gorGEYE

**Actual results**
1-2 line brim around parts surrounded by other parts.

**Expected results**
6mm brim around every part.

**Project file**
N/a, see steps.

**Log file**
N/a

**Additional information**
Seems that the brim lines are limited by widest connection to external perimeter.
When the entrances/exits are tiny, the brim is shrunk accordingly.

64Florian, 2020-05-24:
Hello,
Note for support teams: complete_buildplate_ex_motormount is in Motorized_bogie_v2.zip in the _files_ folder ... 
Adding it for faster debugging
[complete_buildplate_ex_motormount.stl.zip](https://github.com/Ultimaker/Cura/files/4673258/complete_buildplate_ex_motormount.stl.zip)

In your case I noticed you have only one brim on some parts
For the same piece I have a more dense brim:
[image]

-> @viesturz what is the size of your nozzle ? (mine is 0.4mm - I use default "extra fast 0.3mm" )

Your answer could be to disable "Brim only on outside",
but according my tests, this feature seem broken
[image]
[image]
 
@Ghostkeeper  could you confirm if this "Brim only on outside" is broken or not ?
[complete_buildplate_ex_motormount.3mf.zip](https://github.com/Ultimaker/Cura/files/4673283/complete_buildplate_ex_motormount.3mf.zip)

@viesturz For me, the actual brim should be good enough to print your product... 
- Could you try to use a caliper and precisely measure your brim thickness
- According your nozzle size, adjust the thickness of your brim and 1st layer to the half of you nozzle size (I use 0.25 mm 1st layer for a 0.4mm nozzle)
- Check the value of your brim distance
[image]
- Double check the weight of your printed part to see how much you "under extrude" and correct at least the "initial layer flow" and "skirt/brim flow"
[image]

If you all listed points are good, you can increase your build plate temperature, reduce fan speed, and fist of all enclose* your printer if it's not already done... (but for enclose, be careful for proper ventilation of you power supply and the mainboard of your printer)

viesturz, 2020-05-24:
Hey, thanks for the analysis. I have managed to print the piece by using 150% brim line width for better adhesion. Printing in ABS with enclosure.

Have 0.4 nozzle and 0.4 width, my guess the difference in line count is caused by the brim distance of 0.2.

64Florian, 2020-05-24:
> I have managed to print the piece by using 150% brim line width for better adhesion.

Happy to see that !

"For fun" it could be interesting to check the thickness of you brim (just print 1st layer of a product and use a brim 0.25 mm then use a caliper to check if the brim if not 0.35 mm )

But the main thing is you are able to print !

Ghostkeeper, 2020-05-25:
> @Ghostkeeper could you confirm if this "Brim only on outside" is broken or not ?
complete_buildplate_ex_motormount.3mf.zip

It still works for these two rectangles:
[image]

However not for all the small holes. I reckon that's the same as what Viesturz saw with his bug, but then with that it produces 0 lines instead of 1 or 2.

Ghostkeeper, 2020-05-25:
I've seen this result before and always thought that it was sort of nice and probably made by someone who thought hard about the function of the brim. I'll do some digging as to what could cause this.

Ghostkeeper, 2020-05-25:
Yeah, it's caused by this bit here:

https://github.com/Ultimaker/CuraEngine/blob/39bb08c238359d8f1a1fbf7f775b31fc0cc2598a/src/SkirtBrim.cpp#L98-L106

This was [commited 7 years ago](https://github.com/Ultimaker/CuraEngine/commit/70ec7e33ce7d4b0b1080f4240771e29ac17a903e) back when David was maintaining Cura (before my time). It mentions [this issue](https://github.com/daid/LegacyCura/issues/685). It looks like the problem is that otherwise the nozzle ends up going over the same part a lot, which heats up that part and causes overextrusion. This big blob only makes platform adhesion worse.

Ghostkeeper, 2020-05-25:
Right now this is expected behaviour and I don't think it's worth the time to create a setting for it. Nobody is really going to adjust that setting, and the current result is quite reliable. Does anyone have a different opinion?

viesturz, 2020-05-25:
Hi, makes sense to optimize for better adhesion.

However in this case the brim becomes extremely small to the point where it
no longer provides adhesion.

Would it be possible to have a middle ground - draw some reduced number of
perimeters (say 3 or 30% if the brim size) to ensure it does not get down
to a single line.

Viesturs

Ghostkeeper, 2020-05-26:
Well the point is that such small brims don't provide extra adhesion but make adhesion worse. It's still better to have 1 brim line that provides +5 adhesion (arbitrary units, just for intuition) than have two brim lines that provide +5 and -3 adhesion, for a total of only +2 adhesion. You get what I mean?

Maybe one idea would be that it needs to depend on the circumference of the brim line rather than the area. Because the problem is really that the brim line becomes so short that the next brim line goes over the same area before it has solidified.