Ultimaker/Cura #7801: I Can't open Ultimaker Cure.app 4.6.1 on my Mac running OS Catalina 10.15.4

HomeBoxG, 2020-05-21:
**Application version**
Ultimaker_Cura-4.6.1-Darwin

**Platform**
OS Catalina 10.15.4 
Quad-Core Intel Core i5

**Reproduction steps**
1 download the app from Ultimaker
2 dragging it in to the APP folder
3 clicking on the Cure App
4 pop up saying can't open because apple can't check for malicious software. 
[image]

Ultimaker Cura:

  Versie:	4,6
  Afkomstig van:	Ontwikkelaar waarvan de identiteit bekend is
  Laatst gewijzigd:	29/04/2020 11:24
  Ondertekend door:	Developer ID Application: Ultimaker B.V. (V4B3JXRRQS), Developer ID Certification Authority, Apple Root CA
  Locatie:	/Applications/Ultimaker Cura.app
  Soort:	64-bits

64Florian, 2020-05-22:
Hello,
Has I remember you have to "allow" the software in "FireVault"
Sorry my screenshot are in french and on MacOS Sierra but it should be the same 
In application folder: 
[image]
[image]
[image]
On the last screen (after your message only) you should have a button to allow Cura 

Please let me know if you still have this issue I'll try on another OS if it's needed

64Florian, 2020-05-24:
I saw this ticket #6700...
If my last message doesn't help, you may try the @smartavionics build...

Ghostkeeper, 2020-05-25:
Duplicate of https://github.com/Ultimaker/Cura/issues/7389.

The problem is that since Catalina we'd need to sign Cura twice, with both an older signing system and with a newer signing system. Apple isn't very friendly towards its developers. There are ways around it though to still run it (and then you will still run into our Qt version not properly supporting Catalina either, but it runs for the most part).

We're still working on this, but we have been running into account management issues that even Apple Support was baffled about. I quote: "that should never happen" :smile:

closed 2020-05-25 by Ghostkeeper