Ultimaker/Cura #7806: Surface Mode Appears to Override User Seam Location

AndrewAblenas, 2020-05-22:
**Application version**
4.6.0

**Platform**
Windows 10

**Printer**
Creality Ender 5 Plus

**Reproduction steps**
1. Set custom Z Seam Alignment: User Specified, with custom Z Seam X and Z Seam Y.
2. Special Modes - Surface Mode: Surface

**Screenshot(s)**

Shape
[image]

Seam
[image]

**Actual results**
Seam appears to be chosen automatically.  Seam switches location between close edges 2-3 times over the height of the model.

**Expected results**
Seam should choose nearest edge to X and Y coordinates.  I am hoping at the end/beginning of an extruded line.

**Project file**
I will upload after posting this from my phone.

**Log file**
(this doesn't appear to be a bug)

**Additional information**
The model is an stl that contains four separate surfaces (each consisting of multiple faces). 
Each independent surface is separated by about 4mm.

smartavionics, 2020-05-24:
I experimentally made the surface mode polylines heed the seam x/y position and it does what you expect in that it starts the first polyline near to that location. However, there is one layer that still misbehaves so there must be something screwy about the polylines that have been created for that layer.

[image]

Changing the slicing tolerance setting changes which layer(s) are affected so this is possibly an artifact introduced by the conversion of the mesh into slices. The model doesn't appear to have any features near to those layers that would account for the weirdness.

[image]

smartavionics, 2020-05-24:
BTW, these polylines are output in FffGcodeWriter::addMeshOpenPolyLinesToGCode().

smartavionics, 2020-05-24:
It will print as one piece in normal mode with zero infill and no skins...

[image]

Ghostkeeper, 2020-05-25:
> It will print as one piece in normal mode with zero infill and no skins...

I'm getting a normal solid piece. Same as you, but with skin and infill.

Should the seam settings really apply here? It's not a seam since it's not a closed polygon.
Ideally, you'd want to start on one of the endpoints of the polyline, rather than somewhere halfway which happens to be the sharpest corner or the corner closest to a certain location. If you start at one of the endpoints, there won't be a seam at all.

AndrewAblenas, 2020-05-25:
It looks like this issues has been resolved in Burtoogle's (smartavionics?) 20200525 release for Windows-64.  Cura now starts on the endpoint of the line closest to my User-Defined position and will print each line segment consecutively (with no jumping around).  I don't have any rendering artifacts in my slice.  Much appreciated.

I would love to see this in the standard version so I won't close the issue as of yet.

mahtDFR, 2020-05-27:
We've decided to defer this for now as the use case is relatively obscure and we have other priorities, but are happy to accept pull requests.